mnh-jansson/open-battery-information #2: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte

wczajus, 2024-07-23:
Hi, Can not read 'static data', 'battery data' getting the error like below:
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte. (full log attached) 
[error.txt](https://github.com/user-attachments/files/16346834/error.txt)

The UI is populated with values 65.535 and 0.65535 for voltage difference (screen attached) 
[image]

Arduino uno board flashed without any issues log attached ([unoFlash.log](https://github.com/user-attachments/files/16346798/unoFlash.log))

Interface build according schemat attached to the project.

Connected to working battery without any faults BL1380 and BL1840B

Any clue what is worng - I'm guessing it's a communication problem between the interface and the battery because when the interface is disconnected from the battery I get the same error and values ​​in the UI

mnh-jansson, 2024-07-23:
Thanks for the detailed issue.

Could you verify that the batteries are supported?
https://github.com/mnh-jansson/open-battery-information/wiki/Makita#makita-lxt

I'll try to add support for reading battery information for older batteries ASAP. I don't have the unlock for these yet, but with the battery information it is possible to indicate to the user what is possible.

wczajus, 2024-07-24:
The battery what I tested is marked as supported on the Wiki (LIPW014 Microcontroller: MAK02).
I am getting the same error when battery is connected and disconnected from interface.

mnh-jansson, 2024-07-24:
Ok, i think you are right in the guess that the problem is between the battery and the interface. Can you double check the wiring? Please note that the negative battery terminal needs to be connected to GND.

wczajus, 2024-07-24:
I checked the wiring and made new one from scratch. I managed to connect only once.
Reading information about the battery and controlling the LEDs worked for that session.
Then I hit 'Reset errors' and 'Reset battery message' after that I can't connect again. I get the same error as at the beginning.
"  File "c:\Users\XXX\OneDrive\Dokumenty\GitHub\open-battery-information\OpenBatteryInformation\modules\Makita.py", line 65, in on_read_static_click
    model = response[2:9].decode('utf-8')
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte"

mnh-jansson, 2024-07-24:
Is the battery still signaling error with double flashing LEDs? Seems like you definitely have some connection issues between the interface and battery.

Reset errors should be sufficient to get the battery to stop signaling fault. Reset battery message might be needed if the charger is rejected the battery even though the battery itself reports fine.

wczajus, 2024-07-24:
Yes, the battery is still flashing the LEDs.
What should be the connection order or it doesn't matter?
In my case I first connect battery to interface and the power on the Arduino after a while I connect via the UI.
I was also trying to reset Arduino by clicking the reset button on board but it also doesn't help still getting the error and 65 values in UI.

mnh-jansson, 2024-07-24:
The connection order you describe is the same as i use, but it should not matter.

Do you mind including a picture of the interface?

wczajus, 2024-07-24:
Sure, the bread bord just created, both resistors are 4,7k (on the picture one is different but now both are the same).
For the board with SMD, battery minus is directly connected to the Arduino.

[image]
[image]
[image]

mnh-jansson, 2024-07-24:
Very strange, everything seems to be setup correctly. Maybe reduce the amount of connection by connecting GND directly from the arduino to the battery.

I just noticed that the pull-up on my interface for the Onewire is 10kohm, but i doubt that would make a difference. The only other thing i can think of is some dodgy connection on either the breadboard or the makita connector

wczajus, 2024-07-24:
You were right.
The problem was with makita connector. The connector itself works fine as I checked resistance between wire and contact but the problem occurred after connecting to the battery the circuit was broken.
I removed the makita connector and made shortcut directly to the battery connector and got it working.
I have tested the 10kohm on the Onewire and works same way as 4.7kohm.
Thanks for your support!

mnh-jansson, 2024-07-24:
Glad to hear it is working! I'll keep the issue open until i've added correct error handling of connection issues.

mnh-jansson, 2024-07-24:
@wczajus if you have time, please test the latest version. I'm closing this issue for now.

closed 2024-07-24 by mnh-jansson

Marvv89, 2024-07-25:
I'm having the some UTF-8 errors. At first I tried a BL1830 that looks a bit like the LIPW017, but without LED's. 
Next I tried a working (fully assembled BL1850B) also without success, but since charging is still working fine it seems to be very unlikely that the yellow connector is not soldered well enough anymore.

Checked with a scope, sending the messages is going already, but receiving 0xFF's back.
This evening I will also try the new version software to see if there is any progress.