mnh-jansson/open-battery-information #3: OpenBatteryInformation  Debug ERROR??

a9566847, 2024-07-26:
Shown when turned on
Debug: Failed to load interface 'arduino_obi': 'staticmethod' object is not callable
How to solve it?
(Select Interface Field is blank)

mnh-jansson, 2024-07-26:
I'm not able to reproduce. Can you provide more details and which commit you are using?

a9566847, 2024-07-28:
There is a problem with submitting the program code
I have solved the problem
Because 'staticmethod in arduino_obi has multiple symbols @ in front of this line
Delete @ and no error will occur

mnh-jansson, 2024-07-28:
> There is a problem with submitting the program code
> 
> I have solved the problem
> 
> Because 'staticmethod in arduino_obi has multiple symbols @ in front of this line
> 
> Delete @ and no error will occur

I can't reproduce the issue you are describing. You will need provide more details to the issue. Thanks.

a9566847, 2024-07-28:
arduino_obi.py file content
The error in line 6 has not been modified:
import tkinter as tk
from tkinter import ttk
import serial
import serial.tools.list_ports

@staticmethod
def get_display_name():

Revise:
import tkinter as tk
from tkinter import ttk
import serial
import serial.tools.list_ports

staticmethod
def get_display_name():

(After modification, execute PY and it will be normal.)

mnh-jansson, 2024-07-28:
Ok, i think you're using a older version of python and that is the reason why you are getting the error. The modification you are making is not correct, but as the static method is not needed we can just remove it.

Fixed in 836e933ef9830015afd031314d112fa85ff2e820

Closing

closed 2024-07-28 by mnh-jansson