mnh-jansson/open-battery-information #5: An error occurred in MODEL.BL1860 170510-manufacture

a9566847, 2024-08-03:
This version of MCU: F0513+L5203
An error occurred while reading MODEL. Dialog box: Battery is present but not supported.
I don’t know if the board on which you can read the model normally is 1860?
(pcb version number: LIPW012)

mnh-jansson, 2024-08-03:
Have not heard of this PCB revision before. Is it BL1860 or BL1860B? Normally F0513 is not supported but i have been able to read model and voltages from mine. The reset is still unknown.

Could you also provide pictures? 
If possible, front and back such as: https://github.com/mnh-jansson/open-battery-information/wiki/Makita#lipw017

a9566847, 2024-08-03:
> 之前沒聽過這個PCB修訂版。
> 
> [也可以提供](https://github.com/mnh-jansson/open-battery-information/wiki/Makita#lipw017)圖片嗎 ？[](https://github.com/mnh-jansson/open-battery-information/wiki/Makita#lipw017)

[image]

a9566847, 2024-08-03:
[image]
[image]
[image]
[image]

a9566847, 2024-08-03:
[image]

a9566847, 2024-08-03:
[image]

a9566847, 2024-08-03:
[image]

a9566847, 2024-08-03:
> 之前沒聽過這個 PCB 修訂版。是BL1860還是BL1860B？通常不支援 F0513，但我已經能夠從我的設備讀取型號和電壓。重置仍是未知數。
> 
> 還可以提供圖片嗎？ 如果可能的話，正面和背面，例如：https://github.com/mnh-jansson/open-battery-information/wiki/Makita#lipw017

BL1860B (lipw012) and BL1850B (lipw011)

a9566847, 2024-08-03:
> Have not heard of this PCB revision before. Is it BL1860 or BL1860B? Normally F0513 is not supported but i have been able to read model and voltages from mine. The reset is still unknown.
>
> Could you [...]

BL1860B (lipw012) and BL1850B (lipw011)
bl1860b.Opening the wrong screen
[image]
[image]

mnh-jansson, 2024-08-03:
The Arduino OBI reports a strange version, could you try update ArduinoOBI with the latest commit?

a9566847, 2024-08-03:
> The Arduino OBI reports a strange version, could you try update ArduinoOBI with the latest commit?

I am using the latest version you released to compile the exe program

mnh-jansson, 2024-08-03:
Great, thanks for reporting! Sadly the F0513 is still missing unlock, but hopefully you see all the cell voltages.

a9566847, 2024-08-03:
> Great, thanks for reporting! Sadly the F0513 is still missing unlock, but hopefully you see all the cell voltages.

The only unlocking method is to obtain a normal battery from F0513, remove the firmware, and re-program it?....The above is just speculation.

mnh-jansson, 2024-08-03:
The way i've unlocked these is to rewrite a certain block with error counters to 0. The block is at address 0x6800 on the batteries i've extracted the firmware from, but it could differ on yours. If you are willing to risk bricking the batteries i can guide you through the process. In that case i'd prefer if you would contact me through the form on the blog https://martinjansson.netlify.app/posts/makita-battery-post-2

I pushed a commit that displays what i believe is the charge count. Please check it.

a9566847, 2024-08-04:
The number of charges can be displayed, but a new problem was discovered... After resetting the error and message, the indicator light on the battery returned to normal, and the battery damage alarm popped up immediately within 10 seconds of charging the DC18RC (it also popped up again within 10 seconds after unplugging and replugging it) Alarm) Pull out the battery and check if there is any abnormality.

mnh-jansson, 2024-08-04:
Good point. The command to read the battery message triggers the evaluation of the battery, so if no cells are present or any active fault is present it will trigger the lock. 

There must be an active fault, but everything looks good on the screenshot. Could the fuses be blown?

a9566847, 2024-08-04:
> Good point. The command to read the battery message triggers the evaluation of the battery, so if no cells are present or any active fault is present it will trigger the lock.
> 
> There must be an active fault, but everything looks good on the screenshot. Could the fuses be blown?

The HT pin is open circuit, the fuse needs to be replaced first.... Your guess is probably correct.

closed 2024-08-04 by a9566847