mnh-jansson/open-battery-information #6: BL1430 support

pety3bi, 2024-08-04:
As you can see, BL1430 is also can be read, is it possible to make changes for this model, so it can identify only 4 actual cells for this types of batteries?

[image]

mnh-jansson, 2024-08-04:
Thanks for reporting! I'll add another check for the model to adjust to 4 cell. 

If possible, pictures of the BMS would be helpful for the wiki.

nvn887, 2024-08-10:
wrong cell voltage 5 , BL1440
[image]
[image]

mnh-jansson, 2024-08-10:
> wrong cell voltage 5 , BL1440
> 
> [image]
> 
> [image]
> 
> 

Thanks! Can you confirm that unlock works?

nvn887, 2024-08-11:
it's not broken, just reading the wrong voltage.

mnh-jansson, 2024-08-11:
@nvn887 I get that, but i was wondering if you also tried to Clear errors and if that worked.

nvn887, 2024-08-11:
yes, i tried reset (even though it was not an error) but it didn`t work

https://github.com/user-attachments/assets/1a19be25-ee82-4ab7-a82a-ac171711b522