mnh-jansson/open-battery-information #7: BL1840 white board (white fuses) UPD78F0513

MilovanTatic, 2024-08-08:
Hello,
i've managed to wireup one BL1840 board that I have and succesfully read out following data.

Can I somehow contribute to enable clear errors? Anything? Beer? Coffe? xD

Readout:
BL1840
55
20.413
4.125
4.06
4.09
4.072
4.066
0.07
30.31

11 01 0D 02 54 14 09 E9
F1 26 BD 13 14 58 00 00 B1 B1 40 21 D0 80 02 7B 82 D0 8E 67 9F 67 00 01 02 F1 0E 44 00 73 01 31

mnh-jansson, 2024-08-09:
Beer is always welcome 😄.

I hope to have the solution soon. So far i have only been able to unlock by reflashing the firmware which i have unlocked.

MilovanTatic, 2024-08-09:
Well, last night i’ve unlocked a B version. If you want - you could share the fw via email. I’m not afraid to brick it 🤣

p.s. thanks for the work - the B version had only 22 charges and was bricked by a faulty tool.

Check out bymeacoffe :)

mnh-jansson, 2024-08-09:
Much appreciated!!

If you wanna explore BL1840 with UPD78F0513 further, please write to the support email found in the readme

nvn887, 2024-08-10:
wrong cell 5 voltage
[image]
 
[image]

closed 2025-09-25 by mnh-jansson