mnh-jansson/open-battery-information #8: Connector information

eroyee, 2024-08-12:
Martin, 

Well done on your work. I would like to contribute some information from my batteries (if that's useful) but need to know what the 7-pin connector on the battery is so I can order one? 

Apologies in advance if you've said this somewhere but it wasn't immediately clear to me, and the detail wasn't on your graphic. Thought I'd enquire here as I don't imagine I'll be the only one wanting to know :)

Thank you.

Marvv89, 2024-08-12:
Most likely that will be [AliExpress](https://www.aliexpress.us/item/3256804743402583.html?src=google&pdp_npi=4%40dis%21EUR%212.52%212.52%21%21%21%21%21%40%2112000031056585092%21ppc%21%21%21&src=google&albch=shopping&acnt=708-803-3821&isdl=y&slnk=&plac=&mtctp=&albbt=Google_7_shopping&aff_platform=google&aff_short_key=UneMJZVf&gclsrc=aw.ds&albagn=888888&ds_e_adid=&ds_e_matchtype=&ds_e_device=c&ds_e_network=x&ds_e_product_group_id=&ds_e_product_id=nl1005004929717335&ds_e_product_merchant_id=106617809&ds_e_product_country=NL&ds_e_product_language=nl&ds_e_product_channel=online&ds_e_product_store_id=&ds_url_v=2&albcp=19207327905&albag=&isSmbAutoCall=false&needSmbHouyi=false&gad_source=1&gclid=Cj0KCQjw5ea1BhC6ARIsAEOG5pwDx4y_Gu7uorOEt93wdzo0-xMKSKa3jslJttTWw6VRr5yhnL--CnQaAq0fEALw_wcB&gatewayAdapt=nld2usa4itemAdapt)

mnh-jansson, 2024-08-12:
@eroyee Thanks! The more information we have the better! If you have pictures of the insides that is also helpful. Currently we do not have a way to identify BMS versions so having the pictures of the BMS and the data, maybe we can see some pattern. ROM ID and battery message are the most useful for this.

The latter of the ones that @Marvv89 linked is the one i use. If you have the battery open you can just poke some wire in the back of the connector if you don't wanna wait for the connector to arrive. Good luck and let us know how it goes!

eroyee, 2024-08-12:
Thanks to both of you. 

I'd looked on Ali, but wondered if this was a 'standard' connector mildly adapted by Makita, but it appears not (sigh). Will order a couple for the cause, however it may take a few weeks to this location.

In the meantime, for anyone coming here, the pins appear to be standard 0.1" (2.54mm) spacing, and I seem to recall a similar board edge connector in something I've worked on in the past. If I recall what that is I'll report back. 

Otherwise will see what I can do re BMS ID pictures, and I have some other information I'll post separately to this as it doesn't relate to connectors.

aldras, 2024-11-10:
This reply is a little late, but I would like to add a recommendation, which is what I actually did.
Order the cheapest knock off (made in China?) charger you can find (off Amazon or similar) and then dismantle it to recover the connector.
This will not only give you that special yellow connector, but a stable trouble free locking interface to connect with the battery.