mnh-jansson/open-battery-information #11: BL1814N data

eroyee, 2024-08-13:
Image of a BL1815N battery. Board date is 2014, battery serial number starts with 19.

[image]

eroyee, 2024-08-13:
ROM ID: 13 09 04 64 32 03 7E 2F
Batt Msg: 41 43 CB 95 1A 68 00 00 C1 C1 40 41 01 E0 02 03 F0 D0 8E BE 5F 58 00 21 E1 02 0E 01 00 F0 01 03

Notes: This battery is dead from an operational perspective, the 'pack voltage' is 0.079. It was left for a number of years and has a total charge count of 15, it is in a locked state. Battery data read at a similar time to my [BL1840](https://github.com/mnh-jansson/open-battery-information/issues/12) with the temp 1 sensor showing '28.81', and sensor 2 '29.09' with no use and an ambient temp of ~14C.

mnh-jansson, 2024-08-16:
@eroyee  was this battery reported as limited by the software? I currently dont clear the data, so if you have read a fully supported battery and then one with limits, the old value remains.

eroyee, 2024-08-16:
By limited, do you mean 'locked'? As I recall this one reported as locked, but there was no other message, also pretty sure this was the battery I read first (being the least 'valuable' from an operational perspective I figured if I was going to wreck one this'd be it :). Anyway that's useful information, thanks, I'd looked a bit at the Arduino code, but not so much the Python; have been thinking I might do something with a Pico...

rusovich1999, 2024-12-19:
https://github.com/mnh-jansson
Ho i have a bl 1815 is suporte?
and the scheme conection is the battery opened  ?