mnh-jansson/open-battery-information #12: BL1840 date

eroyee, 2024-08-13:
Board image from BL1840 battery (no evident suffix). Serial starts with 14.

[image]

mnh-jansson, 2024-08-13:
Can you add the full ROM ID and battery message to these issues? Thanks!

eroyee, 2024-08-13:
Will do, once I have a chance to make up the interface, just getting the images up for a start. Not sure if the 1815 will return anything yet (very dead), but I have a number of other batteries (1830B's and 1850B's) that should, just can't dismantle to photograph them at this point as they're under warranty...

eroyee, 2024-08-13:
Ok, works well with a Nano for the interface, here's the info for this battery:

ROM ID: 0E 0A 1F 02 03 2B 06 39
Batt Msg: 50 28 ED 83 10 96 00 00 B1 B1 71 41 A2 71 01 5B 42 E0 8E 47 50 5A 00 43 02 02 2E 27 10 7B 01 17

tomaxsas, 2024-08-21:
also has same battery, was drifted voltage, now its locked.
BL1840
48
UNLOCKED
67
17.209
3.322
3.497
3.466
3.408
3.516
0.19
30.01

10 03 03 02 51 C9 04 67
F1 26 BD 13 14 58 00 00 B1 B1 40 21 D0 80 02 3B 82 D0 8E 67 60 5B 00 23 02 E1 0E D3 00 03 00 25

can something be done to add unlocking to it?

aldras, 2024-10-11:
@tomaxsas Have you been able to resolve this?
I think it might have been better to open a new thread for this but I would like to add a comment here based on my experience.

I would not attempt to charge this battery pack again (on the official charger) until the cells have been balanced.
I think what you have experienced is just a temporary rejection from the charger, no hard LOCKOUT as of yet.
If the charger sees this unbalance the pack may be LOCKED out, like what happened to me.

If the battery pack has indeed not been LOCKED out yet I believe you can manually balance (charge) the cells to bring the pack back into specification.  I don't know the balance specification but I think 0.19v is too much.  Someone correct me if I am wrong.
The cell voltages are still above 3.0v so I do believe the cells are ok but I can't see the physical condition of them.

There is some risk (fire, bursting, explosion) to manually charging the cells (if not done carefully), but if you have access to a current limited power supply it should be safe to do.

I performed this on a BL1840 pack last night that is LOCKED out (to keep the cells in good condition) and it took around 45min per cell group to charge (there are 5 groups).  My pack was around 3.5v-3.65v per cell group.  It may take about an hour per cell group to do this, but you may be able to rebalance this pack to bring it back into specification.

After doing some reading up and searching on how to properly charge li-ion battery cells (specifically 18650 type) let me know and we can go into more detail on how to proceed.

tomaxsas, 2024-10-15:
i have balanced it, but battery indicator blinks reds, still no charge. current battery msg:
```
BL1840
48
UNLOCKED
67
20.332
4.072
4.07
4.068
4.065
4.057
0.01
29.84

10 03 03 02 51 C9 04 67
F1 26 BD 13 14 58 00 00 B1 B1 40 21 D0 80 02 3B 82 D0 8E 67 60 5B 00 23 02 E1 0E D3 00 03 00 25
```

ch9511, 2024-12-07:
I have same problem with 2 BL1860B battery, when i putting on charger not charging just red light blinking. After changing temperature sensor one of them started to charge. I dont know about BL 1840 but BL 1860B have 2 temperature sensors.