mnh-jansson/open-battery-information #13: BL1830B data

eroyee, 2024-08-13:
Battery 1, S/N begins 16:

ROM: 10 05 0A 02 07 11 08 A3
MSG: F1 26 BD 13 14 58 00 00 12 12 40 21 D0 80 02 0B C1 D0 8E 67 60 DF 00 03 02 02 0E D2 00 B2 00 E3

Battery 2, S/N begins 21:

ROM: 15 02 08 64 18 08 0D A8
MSG: F1 26 BD 13 14 58 00 00 32 32 40 21 D0 80 02 0E E1 D0 8E A5 60 16 00 03 F1 02 0E 33 00 23 00 B3

Battery 3, S/N begins 22:

ROM: 16 05 1E 64 14 08 06 86
MSG: F1 26 BD 13 14 58 00 00 22 22 40 21 D0 80 02 0C E1 D0 8E 67 60 F2 00 01 02 02 0E 00 00 00 00 21

eroyee, 2024-08-14:
I have one other BL1830B. The board was damaged by a faulty DHP481 (grrr!) which I replaced with a board out of China. This battery returns no data upon request, however it charges and works fine.

Pretty sure I retained the board to take the yellow connector off so I could replace the broken one in my BL1840, however I can't find it at present. If/when I do I'll photograph it and see what, if anything, is readable.

mnh-jansson, 2024-08-16:
@eroyee Yep, the chinese boards only copy the bare minimum of the protocol. I would assume it would give you the battery message atleast? Or maybe they are more timing sensitive.

eroyee, 2024-08-16:
There was no apparent message. At present the connections I'm using to the batteries are a bit 'delicate', but when the correct plug arrives I'll spend some time with this to see what the data stream looks like on the 'scope. You could well be right re timing, although it's interesting that the charger I use with this battery doesn't complain.

Oliver0804, 2024-10-26:
I purchased a BL1830 model battery from China, and I hope the following information will be helpful to you.

>> 33 AA 00
<< FF FF E5 EF CC 7E 3F 29 4A A4 6C E6 E6 92 B6 B4 52 3D A4 A6 94 96 5E 7C B4 FB F4 75 C8 25 7F A4 2C A4 A4 9A 52 4F 4A 24
>> CC DC 0C
<< FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
Attempt 1/2 failed: Invalid response: all bytes are 0xFF
>> CC DC 0C
<< FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
Attempt 2/2 failed: Invalid response: all bytes are 0xFF
>> 31
<< FF FF
Attempt 1/2 failed: Invalid response: all bytes are 0xFF
>> 31
<< FF FF
Attempt 2/2 failed: Invalid response: all bytes are 0xFF
>> CC D7 00 00 FF
<< F7 2C 6C 8A B7 48 4A 94 6C 66 6B A4 2D 26 EC A4 DC 44 34 7A 74 B4 F7 F4 E7 4A 2E FC 53

Talamis, 2025-01-04:
Battery S/N begins 19:
Rom ID: 13 04 0F 64 18 08 01 C9
Battery message: F1 26 BD 13 14 58 00 00 64 64 40 21 D0 80 02 1E E1 D0 8E A5 60 11 00 43 02 02 0E 12 00 B1 00 17

Is Unlocked now, but Charger wont Accept it to Charge and will Flash Red/Green
Cells are in Good health at 33 Charges, Can Draw Lots of Power on Electronic Load and Cells are Balanced within 25mV

Has Anyone tried to decypher the Battery message?

mnh-jansson, 2025-01-05:
Does the charger instantly reject the battery? @Talamis

Talamis, 2025-01-05:
> Does the charger instantly reject the battery? @Talamis

Thanks
It Behaves like charging, with melody for a few seconds, without supplyig power to the pack, then goes into failure Red/Green on Charger.

mnh-jansson, 2025-01-05:
Ok, so then there is a hardware restriction. First step would be to confirm fuse is intact. Other failures may be temperature sensor or cell voltage issues, these usually show up as blinking red only.

Talamis, 2025-01-06:
fuse is ok as i can use the battery in a tool
both sensors did show up as ~28°C
cell voltage is quite close at 3.6V +-20mV

Thanks, now i suspect the charging is shot somewhere on the battery
can wake up the BMS and it sits 0.6V lower than main terminals.

MayDayCom, 2025-02-05:
BL1830B data
15 0B 0C 64 14 08 09 5A
F1 26 BD 13 14 58 00 00 B1 B1 40 21 D0 80 02 0E E1 D0 8E 67 60 F4 00 01 02 02 0E 50 00 50 00 C1

LukaGitH, 2025-05-18:
> fuse is ok as i can use the battery in a tool both sensors did show up as ~28°C cell voltage is quite close at 3.6V +-20mV
> 
> Thanks, now i suspect the charging is shot somewhere on the battery can wake up the BMS and it sits 0.6V lower than main terminals.
> 
> Edit: its the Charge Fuse that has Blown and was only held on by the silicone coating. -> desolderd and bridged Thank you very much!

I have the same situation, A5 status, but: I can charge the battery on terminals with external supply. So which fuse are we talking about than?