mnh-jansson/open-battery-information #16: BL1850B LOCKED?

a9566847, 2024-08-15:
Replace and check that the voltage of each battery string is normal...the fuse is also normal! But it cannot be charged...can it be released?
[image]
[image]

MeatEasterPearls, 2024-09-12:
Can you open the battery and check which microcontroller it uses? Maybe just open it up, take pictures and post them here.

a9566847, 2024-09-12:
[image]

a9566847, 2024-09-12:
The lock status value is 3, the so-called strikeout?

MeatEasterPearls, 2024-09-13:
The microcontroller is a STM32L, which is good. You should be able to unlock it.
What happens when you press "clear errors"?

mnh-jansson, 2024-09-13:
I've discussed this over email with Joe and i suspect it is locked due to the battery message being locked. In its current state, the reset battery message is broken. It's actually good that it is broken because it would do more harm if it would work at this time. I have a solution for this but i have not yet had time to implement it. All time has gone towards F0513 unlock.

Some more details about the battery message. The battery message is the original implementation of the Makita communication protocol and is common through all battery revisions. The earliest batteries were locked by the charger by setting a certain nibble to a non-zero value. This had to be implemented in all the newer batteries so that the charger could be compatible, but the additional commands vary between the newer batteries. The BTC04 even struggles with F0513 and STM/MAK/RAJ and detecting what commands are supported and uses bruteforce to just test and conclude depending on answers. I do the same here. In newer batteries the battery itself has some control of the lock nibble, but the original implementation seems to be present still and can disallow the charger to continue. This is usually the case if the LED indicator on the battery shows no error, but the charger does.

As to if this is related to the "3 strikes and you are out" rumor, no it is not.

Y13Lab, 2024-11-18:
What BMS microcircuit is used on boards with STM32 ?

a9566847, 2024-11-18:
STM32 has integrated BMS circuit

MikaistPilot, 2024-11-25:
Hi, I have many locked boards with the F0513. If you need anything, let me know

Alexys-seven, 2024-12-02:
Hi, I have the same situation with one BL1850B. I have successfully unlocked it once, but there was BMS defect, it show wrong voltage on cell 3 and 4, I have found broken 150 ohm resistor and replaced it (marked red on photo). I think that due to wrong voltage reading, BMS tried to blow fuses for second time (was little bit smoke), after this I can't unlock it anymore. 
[image]
[image]

a9566847, 2024-12-03:
Your current battery status is the same as mine
They should all be locked via battery messages.
Cannot be saved by simple error clearing
The fault pattern is that the charger incorrectly starts blocking three times.
The function of clearing battery information currently has no effect.
This problem requires jansson's assistance
I am also looking forward to the new version to save the battery

khoa300697, 2025-09-18:
chào bạn, hiện tại tôi đang gặp trường hợp này, có thể đọc được model của pin nhưng không thể đọc được dữ liệu Pin makita bl1850. xin hãy giúp tôi khắc phục. thank

[image]

Imported module: makita_lxt
Imported interface: arduino_obi
>> 01
<< 
>> 01
<< 
>> 01
<< 00 02 01
Opened serial port: COM4
>> 33 AA 00
<< 13 0B 18 02 01 3B 0D D9 F1 26 BD 13 14 58 00 00 73 73 40 21 D0 80 02 4B 23 D0 8E 67 63 7E 00 61 02 02 1E 11 00 ED 00 07
>> CC DC 0C
<< FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
Attempt 1/2 failed: Invalid response: all bytes are 0xFF
>> CC DC 0C
<< FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
Attempt 2/2 failed: Invalid response: all bytes are 0xFF
>> 31
<< 18 50
>> CC F0 00
<< 
>> CC F0 00
<< 
>> CC F0 00
<< 
>> CC 31
<< FF FF
Attempt 1/2 failed: Invalid response: all bytes are 0xFF
>> CC 31
<< FF FF
Attempt 2/2 failed: Invalid response: all bytes are 0xFF

bachlonghuynh-hash, 2025-12-26:
tôi cũng gặp vấn đề tương tự bạn khoa