mnh-jansson/open-battery-information #17: BL1415 support

Lectere, 2024-09-10:
Hi there, great project!, 

Wondering if there is support for the BL1415? I think the PCB for this battery is a bit smaller, but they work on the same charger as the BL18** batteries. 

So I cannot imagine it being totally different from the BL18xx batteries.

I also think that the BL1415 has the same board as the BL1430, BL1440, BL1445, BL1460. 

[image]

I've placed new cell's in mine, and since that it won't charge anymore... :(

mnh-jansson, 2024-09-11:
Hey, thanks! This board looks similar to LIPW5003 which is based on MC908 microcontroller. I have very limited experience with these  since i dont have access to very many. I have some idea on how to unlock them, please email me [openbatteryinformation@gmail.com](mailto:openbatteryinformation@gmail.com) and we can investigate further.

a9566847, 2024-09-12:
This looks to be an older model (I have the BL1460B protection board)

Systm21, 2024-12-07:
I have the Same Battery with PCB here but only get "no response". Do you have a fix for that, @mnh-jansson ?

mnh-jansson, 2024-12-27:
It seems that these batteries need an extra GND connection to pin 1 connector of the Makita connector (or just use GND on pin 1 instead of terminal GND). These are not supported, and i don't think i will provide the solution for them as the hardware is quite bad and unlocking them can cause harm because the BMS cannot control individual cell voltages.

In these cases i would almost suggest a good quality BMS replacement, however i don't have alot of good experience with the aftermarket replacements so i cannot give any recommendations there.

Alexys-seven, 2024-12-27:
Be careful with connecting MC908JK3E based BMS to arduino board. Those BMS use enable pin (pin number 6) as power supply. Arduino output can be too weak to supply it, I have burned one of my arduino output when tried to read battery message. Also GND pin is pin number 1 of yellow connector.

AntoneNovikov, 2025-02-14:
I have the same battery BL1415, after replacing the cells it blocked. Arduino connected, but "battery is present but not supported"
I connected the ground to the first pin of the yellow connector as advised @Alexys-seven 
Maybe it is possible to unlock the BMS @mnh-jansson ?

[image]