mnh-jansson/open-battery-information #18: ESP8266 or similar instead of arduino UNO

mekket, 2024-10-15:
Can ESP8266 be used instead if the pin mapping was correct or will it only work with an arduino uno?

mekket, 2024-10-16:
It seems to be working however i have not really been able to connect to any of my "locked" batteries execpt 1.

I will paste the info of every battery i have managed to connect to if this info is useful as right now all of them say that they are unsupported and only for diagnostics.

Working battery, only diagnostics available according to GUI
BL1840
224
UNLOCKED
47
11.553
0.0
0.0
3.873
3.843
3.837
3.87
28.93

0F 09 11 03 56 CE 02 BE
50 28 ED 83 10 96 00 00 13 13 71 41 A2 71 01 5B 42 E0 8E 47 50 5A 00 43 02 02 0E 0E 00 FB 00 A7

Working battery
BL1830 - battery is present but unsupported
760
UNLOCKED
76
0C 0B 1C 65 10 05 11 49
30 26 A1 42 1F 58 00 00 03 03 32 61 23 42 01 5B C1 D0 8E 76 90 DE 00 A3 02 02 2E 8F 20 CE 00 7D

Working battery actual model is BL1840B but also only diagnostics available according to GUI
BL1840
52
UNLOCKED
67
18.378999999999998
3.678
3.675
3.674
3.675
3.677
0.0
29.45

11 04 01 02 54 14 02 1E
F1 26 BD 13 14 58 00 00 94 94 40 21 D0 80 02 1B 82 D0 8E 67 60 A4 00 03 02 02 0E 43 00 D2 00 83

Broken battery flashing left-right on the indicator, actual model is BL1820B and is unsupported sadly.
BL1820
202
LOCKED
67
17.421
3.485
3.482
3.483
3.484
3.487
0.0
29.66

10 0A 18 02 53 CB 03 02
D1 22 BD B3 16 58 00 00 24 24 40 61 B2 81 01 63 41 D0 8E 67 C1 98 00 43 02 02 0E AC 00 9B 00 C7

These are all that i have been able to connect to so far. I will add further outputs if i manage to find and connect to some more if this info is useful.

mekket, 2024-10-16:
Bl1850B, flashing left to right all cell voltages are actually 3.80 but for some reason it is reporting incorrectly. Only diagnostics supported so I cant unlock
BL1850
177
LOCKED
67
16.277
3.803
0.465
5.995
0.017
5.997
5.98
29.62

11 0A 0B 02 54 14 03 19
F1 26 BD 13 14 58 00 00 A3 A3 40 21 D0 80 02 6A 23 D0 8E 67 63 77 00 23 02 02 0E 1B 00 79 00 E5

mekket, 2024-10-16:
BL1820  Only diagnostics supported.
453
LOCKED
6F
14.809
2.965
2.966
2.961
2.959
2.958
0.01
29.32

0F 01 03 03 01 25 05 49
12 21 BD 13 14 58 00 00 B1 B1 40 11 D0 80 02 0B 41 D0 8E 6F BF 3F 00 23 02 02 1E 5C 10 4A 00 35

mekket, 2024-10-16:
BL1830 Only diagnostics supported
363
LOCKED
44
17.538999999999998
3.509
3.509
3.51
3.508
3.503
0.01
29.58

0D 09 16 02 02 92 03 36
30 26 A1 42 1F 58 00 00 73 73 32 61 23 42 01 6B C1 D0 8E 44 91 72 00 23 02 02 1E B6 10 33 00 B5

mekket, 2024-10-16:
BL1840. This one was in my bad pile but i discovered that all of the batteries needed to be put in the original charger in order to "wake them up" after that i can connect to them and read data. This one just seems to have fixed itself.
360
UNLOCKED
47
19.095
3.818
3.822
3.825
3.817
3.813
0.01
29.58

0F 02 0B 03 03 6D 06 8C
50 28 ED 83 10 96 00 00 73 73 71 41 A2 71 01 6B C0 E0 8E 47 50 B7 00 E3 02 02 1E 86 10 F2 00 31

mekket, 2024-10-16:
Older BL1830. Flashes only red on the charger indicating temp issue of some sort? OBI: battery present but unsupported
846
UNLOCKED
76

0C 0B 15 65 15 5F 03 C7
30 26 A1 42 1F 58 00 00 73 73 32 61 23 42 01 7B C1 D0 8E 76 90 DE 00 41 F1 02 3E E4 20 6A 01 85

mekket, 2024-10-16:
Another old BL1830. Flashes green red on the charger. OBI: battery present but unsupported
373
LOCKED
76

0E 03 08 66 10 06 06 1E
30 26 A1 42 1F 58 00 00 B1 B1 32 61 23 42 01 5B C1 D0 8E 76 6F 2F 00 21 02 02 1E 57 10 45 02 B3

mnh-jansson, 2024-10-19:
Glad to hear it is working. Unsupported means that it is either using F0513 or MC908 microcontroller. I do have a solution for F0513, but it is not ready quite yet.

goxon, 2024-11-18:
BL1830
297
UNLOCKED
65

0A 0B 11 01 01 05 02 DD
30 26 A1 42 1F 58 00 00 C3 C3 32 61 23 42 01 5B C1 D0 8E 65 90 B6 00 82 02 02 1E 92 00 AF 01 8A

goxon, 2024-11-18:
do you get some usefull informations in this codes?
if so i can copy all bat i get in repair shop.