mnh-jansson/open-battery-information #19: Подробная инструкция

kulic23, 2024-10-16:
Привет, спасибо большое вам за проделанную работу, не могли бы вы написать подробную инструкцию по пунктам как работать с файлами для мастеров без опыта программирования, я занимаюсь ремонтом аккумуляторов, но в программировании полный ноль, с подробной инструкцией я бы справился намного быстрее, спасибо большое еще раз

aldras, 2024-11-10:
I think we are getting at a stand alone solution similar to the discussion here.
https://github.com/mnh-jansson/open-battery-information/issues/20

I would actually like to create one myself (that does not require a PC to use) but need to set aside time to do so.
What I envision is something with an LCD screen and buttons to start the battery unlock process, read information, etc.

At this point there are a couple of barriers that make this a little difficult.
1. Connecting into the battery (for this I order a cheap Chinese knock off charger and extract the connector).
2. Wiring up the Arduino and connecting the battery.
3. Setting up a computer with Python (with the correct versions and libraries).

Personally I would like to create a stand alone system when I get some free time and this would eliminate the requirement of Python, but will reduce some of the flexibility of the PC requirement.

To give an idea how to get this going I created a Youtube video, if it is ok to post a link I will provide it below.
Feel free to skip through the videos, the second one is quite long since it includes a start to finish process.

This video is basically a first attempt (quickly connecting a battery)
https://youtu.be/G3iKM-AMJ_Q

These next two are trying to assemble something easier to use (but still requires Python).
https://youtu.be/u1KiuAjq80g
https://youtu.be/guNB3_A2Yd0

mnh-jansson, 2024-11-10:
I'm also working on this. Atleast as a first step i will release the solution for F0513 in this premade solution.

[image]

aldras, 2024-11-10:
This looks really nice, a similar direction I would go.
There is just so much useful information available in those MCUs and be nice to find a nice way to display it all.
Thanks for the upload image.

skyzoner, 2024-11-18:
có thể làm video chi tiết về nạp code cho arduino không

skyzoner, 2024-11-19:
hiết lập máy tính bằng Python (với phiên bản và thư viện phù hợp) có thể hướng dẫn video chi tiết từng bước không

eroyee, 2024-11-19:
> hiết lập máy tính bằng Python (với phiên bản và thư viện phù hợp) có thể hướng dẫn video chi tiết từng bước không

If you're asking what I think then:

(1) Program the Arduino with the code from https://github.com/mnh-jansson/open-battery-information/tree/main/ArduinoOBI/ directory
(2) Connect Arduino to machine (USB) and battery to appropriate pins of Arduino (as shown in https://github.com/mnh-jansson/open-battery-information/tree/main/ArduinoOBI#readme)
(3) Copy the files etc from https://github.com/mnh-jansson/open-battery-information/tree/main/OpenBatteryInformation/ directory to a suitable directory on your machine
(4) Assuming you have Python(3) installed, open a terminal and CD to the directory you put (3) files in
(5) Type 'python3 main.py' in your terminal and you should get the GUI
(6) Select the interface, then select the appropriate USB port for the Arduino connected to the machine
(7) Select the battery module (at this stage 'Makita LXT') and click 'read battery model'...

The above assumes you're using Linux, and I've written some of it from memory so there could be some gaps*. Something similar could well work for M$ products but I'm sorry I can't confirm.

* Please, if you can improve this short 'quick start' description, report back as it could help others.

kulic23, 2024-11-20:
> Я тоже работаю над этим. По крайней мере, в качестве первого шага я выпущу решение для F0513 в этом готовом решении.
> 
> ![Изображение](https://private-user-images.githubusercontent.com/39061161/384700137-7a9f0e0e-8109-4d7a-ac10-91f10dd2234d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIxMjY0MjgsIm5iZiI6MTczMjEyNjEyOCwicGF0aCI6Ii8zOTA2MTE2MS8zODQ3MDAxMzctN2E5ZjBlMGUtODEwOS00ZDdhLWFjMTAtOTFmMTBkZDIyMzRkLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyMFQxODA4NDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04ZDYzMGFlZTljNWE1OTE1ODA5YjExM2QxYTRkZTFhMzU4MGY1ZjA0OTM4YmM3YzgyNjA0NjhjMjY5MTI2YjhlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.No-jYyokGJO_PR1XMFPscvOkC5HyTWbPeEVzyfYH1sM)

Thank you very much for your work, probably me and other people who are not good at programming and working with Python, we will wait for your version of the device and instructions for it🤝

mnh-jansson, 2024-11-20:
Thanks guys. I've added some instructions in 1b2169bdc4d2486939268e443a17a169a0aa383e . Please see if it's sufficient for you. If there are missing instructions, feel free to open a PR. Closing this now.

closed 2024-11-20 by mnh-jansson