mnh-jansson/open-battery-information #20: Standalone solution

TobiasPSP, 2024-10-30:
Awesome work! 
would you have an example on how to read battery parameters directly on the arduino side (converting raw bytes to some meaningful reading) ?
The Gui is great but adds so much additional dependencies and requirements like python. Wouldn’t it be relatively simple to display diagnostics on some tft display directly on the mcu side, essentially creating a portable test tool to carry around?
I am more than happy to create one if you could fill me in on the protocol details.

thx & keep up the great work!

aldras, 2024-11-10:
From my experience so far, the Arduino is more of just an interface that provides protocol conversion (from UART to OneWire and back) and the heavy lifting is done on the Python side.  The Python script assembles the binary strings to send to the battery, and the Arduino just forwards them over the OneWire interface.  Reverse is true when a reply is received.

If you look closely at the Python code you will notice the binary query strings and response parsing.
This is the basis for making a self contained setup, which I personally would also like to do (with an LCD screen) but that project is for another day (for me anyway).

mnh-jansson, 2024-12-27:
Sorry for getting back to this so late. I have a working solution now that is a fully self-contained solution. I doubt i will open-source this anytime soon though.

TobiasPSP, 2024-12-27:
That is awesome, congrats! 

May I ask: are you currently just too busy to do a proper writeup, or are you aiming to monetize your code? Both is perfectly ok.  Just trying to better understand.

Happy New Year!

mnh-jansson, 2024-12-27:
More of a prioritization issue. I'm working hard on figuring out other brands like Hikoki, Makita XGT and Milwaukee. This is the part that interests me the most. I have spent some time updating the READMEs on the Github, not sure if this would suffice as a proper writeup, but i try to update it as i go.

While i'm not really planning to monetize my code, i am interested in finding a way to distribute this solution in a way that is harder for others to sell. And while it does not bother me that people try to sell the project, this was expected i guess, it bothers me more that people are buying it. 

Releasing a standalone/ self-contained solution would definitely contribute more to this "issue". What i plan to do in the future is providing a web-based tool that would require logon. To some degree i want to provide the solution for free, but have people that make a living of it contribute to the evolvement of the tool. In my opinion, diagnostics information should be available to everyone, it's a fundamental part of right to repair. The automative industry has had this requirement for decades and i would say it is generally accepted that everyone should have the right to fix their own car. In my mind there should be no exception with any other product/industry. It's also interesting to see that EVs, these things that are considered to save the planet, are so much harder to repair and diagnostics tools and information is hidden. Obviously this is to make more money and shorten the possible life-span of the product, so how important is this green-movement really?

Did not mean to sound too do-goody or get political, just wanted to explain what i see the purpose of the project to be. Ideally it should not even have to exist.

a9566847, 2024-12-27:
I accidentally found a Chinese website selling your development tools
The language is simplified Chinese. I am considering whether to purchase from a Chinese website.
I don't know if you are authorized to sell it.
Almost the same as the one you developed
[image]

mekket, 2024-12-27:
https://www.tiktok.com/@kangyijie/video/7441914868195872056
https://www.tiktok.com/@kangyijie/video/7437881616443247928
In these videos they are unlocking BL1860B-s so they might not be unlocking all the batteries just yet? The 1860B-s that they show probably don't use F0513 so they might just be using/selling @mnh-janssons project as their own. It sure does look similar even the icon looks the same to OBI in one of the videos when the device starts...

mnh-jansson, 2024-12-28:
It's my work, but it's not stolen. I've made this version as part of an agreement with them. 
If anyone is interested in getting one directly from me, contact the email address in the README.

I'm closing this issue for now as it is in the roadmap of this project.

closed 2024-12-28 by mnh-jansson