mnh-jansson/open-battery-information #21: communication with the charger

Y13Lab, 2024-11-01:
Hello.
Have you studied how the charger communicates with the battery?
I have a homemade battery-powered welding machine and I wanted to make the battery compatible with the Makita charger.
https://www.youtube.com/shorts/GvXqMOTdAs8
But I didn’t succeed, maybe I don’t have enough knowledge or perseverance.

Help as much as you can.

aldras, 2024-11-10:
Assuming you are attempting to make a spot welder.

If it is the 18V batteries you are attempting to use I don't think you will get the current you need from them.
The specifications for the cells in a 18V 4Ah battery pack I have is 2200mA max discharge (two cells in parallel) so this comes to 4400mA maximum discharge current, I don't think the battery pack can safely supply the current you need.

You may or may not have better luck with the 40V batteries but now you are getting into a higher voltage (and bigger spark).

The type of battery that can effortlessly supply this kind of current (hundreds of amps) is a lead acid (car battery) type of battery.  I would recommend a 12V variety, and even a smaller sealed lead acid may work.  Be careful of gas buildup when quickly discharging a sealed battery (there is no vent outside of an emergency pressure relief).
These types of car batteries are designed to supply hundreds of amps while starting a car.

This would be my recommendation instead of a Makita battery pack.
I honestly don't think you will be able to pull the amount of current necessary to spot weld with these.

aldras, 2024-11-10:
I just looked at the video, it is spot welder, but a normal wire feed kind of welder, but the same applies.
Normally a wire fed welder (if I am not mistaken) operates at a very low voltage (maybe 1-5 volts) with very high current (hundreds of amps) since it is the current (not the voltage) that produces the heat to weld.

I still think you can do this with an automotive battery, but be careful of short circuits and I would fuse protect it (50A or 100A fuse, or more).  You will also need a way to feed the wire.

mnh-jansson, 2024-11-14:
Hey @Y13Lab 

Sorry for not replying sooner. I have made a battery emulator that would fit this purpose. I did it mostly to verify my findings, and also to make a automatic tune changer :) But making batteries is not my intent with this project, and releasing such work would most likely just be used to make clones. I would still be willing to help you with your project, but please contact me through the contact e-mail in that case.

Closing this issue now as it is not relevant to this project.

closed 2024-11-14 by mnh-jansson