mnh-jansson/open-battery-information #23: After resetting the BL1850B battery, the red LED on the charger blinks.

MayDayCom, 2024-12-14:
After resetting the BL1850B battery, when charging with the genuine charger, the red LED (temperature abnormality) blinks.

Only the red LED blinks, and it does not charge. There is no problem with the temperature sensor or the temperature.

This seems to be a kind of circuit lock. I wonder if this problem can also be solved.

mnh-jansson, 2024-12-18:
Usually means that the fuse is blown or cell voltages are uneven.

MayDayCom, 2024-12-18:
If you keep resetting with the program you uploaded, it will work normally at some point. Cell balancing is normal and there is no problem with the fuse. If you reset with the program, the above symptoms will occur randomly one by one. If you keep resetting it several times, it will be resolved. I really don't know. If there is any data you need, I will send it to you. Thank you very much for the program.  It may be awkward because it was uploaded using a translator. Thank you.

ch9511, 2024-12-22:
I have seam problem with BL1806B when I put it on charger just red light blinking, but battery indicator show than battery full charged. In other battery with same problem I was find out than thermistor was broken, after changing it, charger started to charge, but charger ventilator not starting. In program shows temperature 25C for sensor 1 and 29C for sensor 2  Some body know what is thermistor nominal for BL1806B battery 10K or 20K ?

a9566847, 2024-12-25:
The thermistor that detects battery temperature is 10K ohms

MayDayCom, 2024-12-26:
Thank you~ 
The strange thing is that even after replacing the thermistor, the same symptoms persist. Sometimes, when resetting with the program, it randomly comes back to life. This is the problem. I can't find where the issue is.

The sentences might be strange because I used a translator.

ch9511, 2024-12-29:
I was played with battery BL1860B, PBC LIPW014 which had problem than red light blinking , but charging and program show than battery is unlocked. After pushing clear errors button battery was locked and charger start to blink red and green , but after pushing clear errors button and after that reset battery message, it was unlocked and started to charge as normal.