mnh-jansson/open-battery-information #27: Does makita LIPW014 not support?

MayDayCom, 2024-12-24:
[image]
[image]
[image]

Does makita LIPW014 not support?

[image]

ch9511, 2024-12-26:
This message popup when wires not connected to battery. Check if all wires connected correct, battery voltage not lower than 19v also you can try NEWAS and TH connect together when you try to read battery.

MayDayCom, 2024-12-26:
Thank you. I will try again.

MayDayCom, 2024-12-27:
Connection successful. Thank you.

mnh-jansson, 2024-12-27:
Closing for now, feel free to reopen if the issue was not resolved.

closed 2024-12-27 by mnh-jansson