mnh-jansson/open-battery-information #30: Flagged as Virus by Windows

Talamis, 2025-01-04:
Tried to download and execute the .exe but its getting removed and flagged as virus.

ch9511, 2025-01-04:
https://github.com/Shampra/EvernoteToAnytype/issues/10

Talamis, 2025-01-04:
That is terrible, alot of people wont be able to even try it.

Badboydas, 2025-01-04:
I am not Linux literate, but am using it on windows.
Install python following this video https://www.youtube.com/watch?v=YKSpANU8jPE&t=204s
Install PIP using this video (but ignore pandas https://www.youtube.com/watch?v=nj8gVMH6yR8.
Then follow themnh-jansson instructions.

mnh-jansson, 2025-01-04:
Thanks for reporting. I will have to look into how to solve this with pyinstaller. I think the precompiled executable is a must for alot of users.

Dmytro2638, 2025-01-06:
> Спробував завантажити та запустити .exe, але його було видалено та позначено як вірус.

Я зробив відео на youtube, як обійти антивірус, і запустив на windows

Tench745, 2025-03-14:
I was able to get the windows executable to run despite being flagged by Windows Defender as a trojan virus.

ch9511, 2025-03-16:
When jumping out defender, just push more info and run anyway. There is not virus just, Windows defender identifies all excitable files as dangerous objects which are not correctly singed.

nbctcp, 2025-09-18:
Norton flagged as virus
it trying to connect to https://release-assets.githubusercontent.com

ChiliApple, 2026-08-15:
I ran into this as well, and since the thread has been open a while I thought I'd share what I found when I looked into it — hopefully useful rather than noise.

Short version: it looks like a reputation-based false positive rather than anything in the code.

File: `OpenBatteryInformation.exe`, SHA256 `16df06b0c76defff5c076d98e6680f52ba61044387db1b18bd4b284eb756e5eb`

VirusTotal shows 27/70, but the split is what matters:

- Clean: Kaspersky, ESET-NOD32, Bitdefender, Emsisoft, Dr.Web, Avira, CrowdStrike Falcon, ClamAV — essentially every strong signature-based engine
- Detecting: almost entirely ML/reputation verdicts (`FileRepPup`, `Generic.ml`, `Static AI - Suspicious PE`, `Mal/Generic-S`, `Trojan.Gen.MBT`, `W32/PossibleThreat`)

Behaviour in the sandboxes is unremarkable too: the only hosts contacted are `www.microsoft.com` and Microsoft/Google IP ranges (all 0/91), there are no IDS rule hits, and every MITRE signature is INFO level only. The file has been on VirusTotal unchanged since 2024-11-25, and the strong engines still consider it clean after that time.

As you suspected in your earlier reply, PyInstaller does look like the trigger: an unsigned bundle with a compiled bootloader and low prevalence is close to a worst case for reputation-based scoring, and it tends to recur with every new release.

I've submitted the file to Microsoft as a false positive (submission ID `019385ff-2d5c-4928-a493-fe6670feeb1e`), which should at least clear this specific hash.

Two things that might help beyond that, entirely your call:

- Publishing SHA256 checksums alongside releases is cheap to do and lets users confirm they have the genuine build. At the moment there's no published reference to compare against, which also makes reports like #101 hard to assess from the outside.
- [SignPath Foundation](https://signpath.org/) offers free code signing for open-source projects. It wouldn't help retroactively, but it would break the cycle for future releases.

Thanks for the tool either way — it does exactly what I needed with my Makita packs.

TobiasPSP, 2026-08-15:
Running the python script directly on Windows imho is a simple and reasonable workaround. 

Running the exe basically does the same thing but since this is also a typical malware attack vector, it gets flagged, and some AV eat the exe altogether. 

Whatever you choose to do, lowering your general security settings, and disabling real-time protection would be the worst „solution“. 

I ended up using Powershell instead of the python script, you can basically choose whatever language suits you best to query the COM interface that is opened by the Arduino when you connect it to a PC.

mnh-jansson, 2026-08-15:
From now on the UI can be found at: https://openbatteryinformation.github.io/, no more false virus alerts.

closed 2026-08-15 by mnh-jansson