mnh-jansson/open-battery-information #31: Battery is present but not supported

ironviet, 2025-01-07:
I have a BL1805 battery. When connected it cannot read the full information. please help
[image]
[image]
[image]
[image]
[image]

mnh-jansson, 2025-01-07:
Could you send a picture of the BMS and possibly any markings that you may find on the board? I dont think i have encountered this BMS myself. If you are would like help support this battery, please email me.

ironviet, 2025-01-07:
hello! i will attach you the picture. the chip name is R2J240

khoa300697, 2025-09-18:
I also have a similar situation; all 3Ah, 4Ah, 5Ah, and 6Ah battery types with different chips can display information. 
Only one type of IC with the battery code 1850B cannot display information.

[image]

F1 26 BD 13 14 58 00 00 73 73 40 21 D0 80 02 4B 23 D0 8E 67 63 7E 00 61 02 02 1E 11 00 ED 00 07

I don't know where I went wrong.

[image]

[image]

[image]

Please help me. :(