mnh-jansson/open-battery-information #33: Repeated battery error BL1850

Vasilii008, 2025-01-12:
Hello, BL1850 battery, the battery error is reset, the battery starts working with power tools, but after charging on the original charger, the battery crashes again.
[image]
[image]
[image]

Vasilii008, 2025-01-12:
[image]
here is information from the program

ch9511, 2025-01-14:
It looks than some cells lost capacity. Battery have almost 400 charge discharge cycles. I can't see what is registered maximal voltage difference on battery, but 0,2-0.3v voltage difference it's starting to be problem for battery and charger locking it.  You can try to balansere cells, if that not help you have to change all cells.

Vasilii008, 2025-01-14:
The most interesting and surprising thing is that after unlocking the battery, it works with power tools, discharges during use, and when I put it on the charger, the charging station charges it, gives a signal about the full charge, but then, when I insert it into the power tool, it becomes locked again. Could it be that there is a limit on the charge-discharge cycle, and the system blocks the battery?

mekket, 2025-01-14:
I have some older batteries that report around 900 cycles and they still work but to me it seems like every time you put the battery on the charger the count goes up even if it is for a short period of time. Someone can correct me if I am mistaken about this.

mnh-jansson, 2025-01-14:
I actually have two batteries with the same issue now. Both are LIPW014.

@mekket It should be the charger that sets the count, atleast this is the case for older batteries. Once the cycle stops, the charger writes a new battery message to the battery.

Vasilii008, 2025-01-14:
@mnh-jansson Thank you for the program; it works perfectly. If you come up with any solutions regarding this issue with this type of battery, I would be grateful if you could share them.

ch9511, 2025-01-14:
Sony Murata vtc6 cells after 500 full charge discharge cycles can lost up to 30% of capacity. I don't find any information like that about Samsung cells, but I think it must be similar. Yes, Makita battery counting every time when you putting battery on charger, also counting how many times battery was overheated, maximal cell voltage difference, how many times battery was over discharged and over charged. from this information it generating something like battery health report. I think battery status code it is that health report, because Makita battery analyser show parameter like battery health status. Charge count, I think can be calculated by connecting TH and NEWAS together.

mnh-jansson, 2025-01-14:
@Vasilii008 

Thanks! I suspect there is some error still present in my BMS atleast, but i have not figured out what it can be yet. I hope to be able to provide the exact cause of the error in the future, but it will take alot of experimenting and reverse engineering.

EisBear88, 2025-09-28:
I also have a BL1850B with the same BMS and the exact same issue. The battery only had about 30 charge cycles when it started happening. 

[image]

Has anyone made any progress regarding this problem? 

(p.s. thanks so much for all the hard work on this!)

Bennypro66, 2026-03-15:
Hello 👋, have you found any solutions?