mnh-jansson/open-battery-information #35: Arduino uno fixed inside an old Makita charger

Orgman67, 2025-01-17:
I watched a couple of videos to gain ideas and thought I would share what I have built to make the process of analysing batteries easier. I removed most of the old electrical components, used a hot glue gun to fix the Arduino plastic mount onto the old PCB. I cut out holes to enable the USB and 5V connections.

[image]
[image]
[image]

mnh-jansson, 2025-01-17:
If you have access to a broken charger it is definitely convenient. Unfortunately not everyone has it, but nice build!

Just a note, you dont need the DC barrel jack connection. The 5V over USB is enough.

MayDayCom, 2025-01-18:
I also used a broken Makita charger and put an Arduino inside. It's really convenient. It's also easy to insert the battery.

MayDayCom, 2025-01-20:
I've saved several and am still working on some. thank you

aldras, 2025-02-06:
That is pretty cool, thanks for sharing.
I did something similar and can share a picture.
Bought a cheap charger off Amazon and used that, ran the USB cable out the hole the power cable was.

I actually used an Adafruit Trinket M0 (as opposed to an Arduino Uno) since it was smaller.
I documented the whole process and hope it can give others ideas.
I am planning to get the Python program working on a Raspberry Pi next (with a touch screen).
It will make a kind of self contained appliance (can I use appliance?).

It is not quite as self contained as having an LCD connected to an MCU directly (and not need a PC) but I think it could be the next best thing.

The step by step process is documented in the two videos below.
Might make a video about the touch screen Raspberry Pi.
https://youtu.be/1yGGF7F5GFM
https://youtu.be/3QhsOQUzpIg

[image]

Orgman67, 2025-02-06:
> That is pretty cool, thanks for sharing. I did something similar and can share a picture. Bought a cheap charger off Amazon and used that, ran the USB cable out the hole the power cable was.
>
> I [...]

I was actually inspired by your work so I should be thanking you for sharing it. 👍 I would be very interested in how you go with a touch screen as this is of interest to me.

mnh-jansson, 2025-02-07:
Thanks for sharing! It's interesting to see what everyone are creating, but i feel like the issue section is not really suitable for it. I opened the Discussions section which should be more suitable.
https://github.com/mnh-jansson/open-battery-information/discussions

closed 2025-02-07 by mnh-jansson