mnh-jansson/open-battery-information #36: BL1820B only in read mode

Bojidar-sir, 2025-01-17:
First I would like to say that this is an amazing project and you guys have done the impossible!
On to the issue - I did create the provided setup as I have quite few if these batteries and one BL1820B recently locked its self, assuming as it was used and charged in very cold environment as the issue occurred at this very moment. 

The battery is from 2018, 2.0 AH, SN# 182Y03FWS - F01715. The tool connects to the battery but it states only diagnostic, though it can read the battery data. 

[image]

Debug info (Imported module: makita_lxt
Imported interface: arduino_obi
>> 01
<< 
>> 01
<< 
>> 01
<< 00 02 01
Opened serial port: COM3
>> 33 AA 00
<< 12 0B 03 02 06 33 06 B3 D1 22 BD B3 16 58 00 00 B1 B1 40 61 B2 81 01 13 41 D0 8E 67 3F B1 00 43 02 E1 0E 17 00 26 00 F7
>> CC DC 0C
<< FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
Attempt 1/2 failed: Invalid response: all bytes are 0xFF
>> CC DC 0C
<< FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
Attempt 2/2 failed: Invalid response: all bytes are 0xFF
>> 31
<< 18 20
>> CC F0 00
<< 
>> CC F0 00
<< 
>> CC F0 00
<< 
>> CC 31
<< 51 0F
>> CC 32
<< 4F 0F
>> CC 33
<< 4F 0F
>> CC 34
<< 4F 0F
>> CC 35
<< 4F 0F
>> CC 52
<< 92 0B
) 

Board PIC 

[image] 
if not clear I have try and send another one.

I have another 2ah - SN# 212315GWS battery that is from 2021 which is not locked, and goes as far as the above one and the results look similar in the tool.

[image]

Debug info (Imported module: makita_lxt
Imported interface: arduino_obi
>> 01
<< 
>> 01
<< 
>> 01
<< 00 02 01
Opened serial port: COM3
>> 33 AA 00
<< 15 03 0F 02 08 27 00 96 D1 22 BD B3 16 58 00 00 94 94 40 61 B2 81 01 13 41 D0 8E 67 C0 50 00 23 02 02 0E F3 00 C3 00 35
>> CC DC 0C
<< FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
Attempt 1/2 failed: Invalid response: all bytes are 0xFF
>> CC DC 0C
<< FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
Attempt 2/2 failed: Invalid response: all bytes are 0xFF
>> 31
<< 18 20
>> CC F0 00
<< 
>> CC F0 00
<< 
>> CC F0 00
<< 
>> CC 31
<< EF 0E
>> CC 32
<< EE 0E
>> CC 33
<< ED 0E
>> CC 34
<< EE 0E
>> CC 35
<< EE 0E
>> CC 52
<< 78 0B
)

One strange this that I found is that all my 5am batteries can use all the options from the tool and not only diagnostics.
for example i have one 5ah - SN# 188921HV also from 2018, which is the same generation as the above locked 2.0ah battery (My assumption based on the year) and this 5ah battery has no issue to use the entire spec from the tool, moreover to be cleared of errors.

[image]
debug info (Imported module: makita_lxt
Imported interface: arduino_obi
>> 01
<< 
>> 01
<< 
>> 01
<< 00 02 01
Opened serial port: COM3
>> 33 AA 00
<< 12 09 16 64 05 05 1E 81 F1 36 B6 C3 18 58 00 00 84 84 40 21 01 80 02 5B 43 D0 8E 1B F0 6A 00 63 02 02 0E 67 00 56 00 A9
>> CC DC 0C
<< 42 4C 31 38 35 30 42 00 00 00 00 00 00 00 00 00
>> CC D7 00 00 FF
<< 09 4E 9B 0F 9B 0F 96 0F 9A 0F 9C 0F 23 05 75 0B 7C 0B 00 80 40 B9 52 35 11 FE 2D A0 00
).

Please let me know if an additional information is required for me to try and help further with the project.

mnh-jansson, 2025-01-17:
Thanks, i think i have this solved. If you are interested, we can do a remote session and i can try to unlock this one for you. It would be interesting to extract some information from the working one as well if the unlock does not work. Contact the email in the readme in that case.

As for 2Ah and 5Ah in same year having different BMS, that's very possible. Why the commands changed, i have no idea, but it's a pain for sure.

Bojidar-sir, 2025-01-18:
Hi Martin

That would be great. I am free on Sunday and we can meet from 3pm Eastern
European time to 11:59pm if that works with you?
Once again, thanks for the support as you guys are doing an amazing job
helping people with this project.

Bojidar-sir, 2025-01-19:
@mnh-jansson 
Hi Martin, 
I am leaving my email in the thread in case you have availability ( Bojidar.Sir@gmail.com )

woyalishanda, 2025-04-08:
Have you managed to fix it, my friend?

amensparta, 2025-04-10:
i just try the open battery information. from what i found is, the battery BL1840 can be reset using the current onewire(tools)obi.exe. but i am stuck with 5 piece BL1820B board that i am unable to reset, just read data currently. any help with this would be awesome and save the board for another day. thanks

nectogale, 2025-05-26:
Hi there, a little UP about this topic, because I have the same issue with a Makita BL1830.

ROM ID : 10 02 0C 02 03 19 07 8E

 Here's debug infos :

> Imported module: makita_lxt
Imported interface: arduino_obi
>> 01
<< 
>> 01
<< 
>> 01
<< 00 02 01
Opened serial port: /dev/ttyACM0
>> 33 AA 00
<< 10 02 0C 02 03 19 07 8E F1 26 BD 13 14 58 00 00 43 43 40 21 D0 80 02 6B C1 D0 8E 67 65 D2 00 42 02 02 0E 07 00 76 00 66
>> CC DC 0C
<< FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
Attempt 1/2 failed: Invalid response: all bytes are 0xFF
>> CC DC 0C
<< FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
Attempt 2/2 failed: Invalid response: all bytes are 0xFF
>> 31
<< 18 30
>> CC F0 00
<< 
>> CC F0 00
<< 
>> CC F0 00
<< 
>> CC 31
<< 3E 0E
>> CC 32
<< 3D 0E
>> CC 33
<< 3B 0E
>> CC 34
<< 3D 0E
>> CC 35
<< 3E 0E
>> CC 52
<< 6B 0B

derivitiv, 2025-10-06:
Hello,
I also have this battery and I am experiencing the same behavior. The sad part is that the battery reports only 6 charges before it locked. Is there any movement on this issue? Will there be an update to the project in the next few months?

muslim5630, 2026-02-20:
Hello! The problem is still unresolved?