mnh-jansson/open-battery-information #37: Arduino uno fixed inside an old Makita charger

Orgman67, 2025-01-20:
Help needed.
I have a bL1860B battery that had a broken nickel on battery 1. I have repaired this and the program is reading the voltages correctly. I have physically measured the voltages and they line up with what the prgram is reporting refer 1st screen shot. 

I have reset the battery and what I have found is that is that if I clear the errors and reset the battery message the battery stays locked.

If I clear the the errors and don't reset the battery message the battery is unlocked. Refer Screen shot 2. 

I place the battery on the charger and it starts to charge and then after a few seconds the charger stops and indicates a faulty battery.

[image]

[image]

mnh-jansson, 2025-01-20:
Need more information about the BMS version. Suspect it is same issue as #33 .

Orgman67, 2025-01-20:
Yes my battery is also LIPW014.

[image]

Orgman67, 2025-01-20:
> Need more information about the BMS version. Suspect it is same issue as [#33](https://github.com/mnh-jansson/open-battery-information/issues/33) .

Yes my battery is LIPW014
[image]

I'm also finding the same with the following battery which is BL1850B
[image]

Bennypro66, 2025-01-22:
[image]

Orgman67, 2025-01-22:
> Hello orgman67.. following an error trigger following a cut in the nickel strip on the bl1860b model the fuse automatically commits suicide. if you repair it it will charge again

Thanks so much I will definitely give this a go and will let you know how I go.

Orgman67, 2025-01-24:
> > Hello orgman67.. following an error trigger following a cut in the nickel strip on the bl1860b model the fuse automatically commits suicide. if you repair it it will charge again
> 
> Thanks so much I will definitely give this a go and will let you know how I go.

Ok I've had success using the tool to recover a locked BL1860B battery. This particular one had a good fuse but had 2 faulty cells. I replaced the cells, rebalanced the voltages and after a couple of goes managed to get the tool to unlock and recover the battery. 

What I found is if I clear the errors, the battery would unlock, but as soon as I tried to reset the battery message, the battery would lock again. When cleared the errors but didn't reset the battery message, this worked for me.

Orgman67, 2025-02-06:
> > Hello orgman67.. following an error trigger following a cut in the nickel strip on the bl1860b model the fuse automatically commits suicide. if you repair it it will charge again
> 
> Thanks so much I will definitely give this a go and will let you know how I go.

Hi Bennypro66, I got my new fuses yesterday and changed them on a couple of batteries and they now work!!!!!. Thanks for helping me to save these batteries from collecting further dust.

Orgman67, 2025-02-06:
> To add a little to this, this is expected behavior when a fault is still present. Part of the process of resetting the message will trigger the battery management MCU to perform a check and at that time will set the lockout flag if the battery fails the fault check. I suspected that the battery would automatically relock if there was a cell balancing issue, and in this case it happened. A cell misbalance is one of the many faults that the MCU checks for, and I believe setting the lockout flag was good in this case (a real fault). I posted a video on YouTube showing a method to manually charge and rebalance the cell banks (if that helps anyone down the line). That video can be found here https://youtu.be/4ZcgfOGrU10 Glad that replacing the cells fixed the battery, great job!

Hi aldras, Makita must have a fairly strict tolerance level for cell unbalances as even this battery goes into lock mode after I clear the errors and then reset the battery message. If I clear the errors and I don't reset the battery message the battery remains unlocked. I'm not sure if I am doing the wrong thing or not.

[image]

Bennypro66, 2025-02-06:
> > > Bonjour orgman67.. suite à une erreur de déclenchement suite à une coupure de la bande de nickel sur le modèle bl1860b le fusible se suicide automatiquement. si vous le réparez il se rechargera à nouveau
> > 
> > 
> > Merci beaucoup, je vais certainement essayer et je vous ferai savoir comment j'y suis parvenu.
> 
> Salut Bennypro66, j'ai reçu mes nouveaux fusibles hier et je les ai changés sur quelques batteries et elles fonctionnent maintenant !!!!!. Merci de m'avoir aidé à empêcher ces batteries de prendre davantage de poussière.

Bonjour orgman67 
Parfait 👌 très content d'avoir aidé 💪

aldras, 2025-02-07:
I don't think there is anything wrong with the way you are doing it.
I don't see any thing wrong with attempting resets multiple times, there is really nothing that can wear out except maybe EEPROM memory, but that has thousands of writes, nothing to worry about.
Yeah, Makita could be a little more strict, and could vary between model as well.
All depends on what the engineers that designed that specific model decided to do I guess.
I do have a battery that is slightly misbalanced by like 0.05 or so I think, it is an older battery.
Great work on getting those recovered though.

mnh-jansson, 2025-02-08:
Closing this as it seems to be resolved. These kinds of issues are better suitable for the newly opened discussion section and from there could ultimately end of in the Wiki so we can create better guidance for everyone.

closed 2025-02-08 by mnh-jansson

Tromppa, 2025-10-26:
Hey @Orgman67 I have exactly similar problem with the fuse gone, and it not charging after unlocking. Where did you order a new fuse? Or how did you find the type of fuse it needs. I also have a LIPW014 board. 
And @mnh-jansson this is an awesome tool you've built, thank you!