mnh-jansson/open-battery-information #40: BL1830 info

TBAMax, 2025-02-11:
BL1830 with F0513 microcontroller. Connects nicely and detects battery type correctly. Reports unlocked, but voltage measurement readings are absolutely off. Do not start charging on CD18RC original charger-- flashes red-green immediately.
When I got it one cell pair was short circuit (in the cells), faulty cell replaced (to try out if it will charge), and pack balance charged to 3.75V per cell. 
This board do not have special BMS chip for voltage measurements. Instead it have only one LM324 quad opamp(on top side of PCB), so the guess is it is probably used for individual cell voltage measurements. Microcontroller is located on the bottom side of the board.
The board do not have any built in undervoltage protection. Batt+ and Batt- connected directly to the output terminals.  TH pin (that some tools may use for enable battery) is directly connected to the Batt+ via the heated fuse (that can only  be blown once by the circuit (was not blown in my case)). 

[image]

BL1830
64
UNLOCKED
65
69.367
0.022
9.814
45.076
11.382
3.073
45.05
20.77

0A 02 1C 02 01 10 02 E9
30 26 A1 42 1F 58 00 00 B1 B1 32 61 23 42 01 2B C1 D0 8E 65 90 42 00 43 F1 C1 0E 04 00 73 00 97

[image]

[image]

[image]

mnh-jansson, 2025-02-16:
Very strange measurements! Was the board connected when you did them? LIPW002 and LIPW003 are two boards i used alot in the development an had no issues with the cell voltages.

TBAMax, 2025-02-16:
> Very strange measurements! Was the board connected when you did them? LIPW002 and LIPW003 are two boards i used alot in the development an had no issues with the cell voltages.

Thanks for the info. Board was connected when the measurements were as shown. It is possible there is some other(physical or component) fault with the board. Now that I know that it is quite commonly used board, I can have a closer look.

samakinen, 2025-03-28:
I have a BL1830 with similarly looking board and very similar voltage readings as @TBAMax. The battery working and is charging fine in DC18RC charger, so I assume the BMS is working as expected.

Here is the data I got from my battery:
BL1830
28
UNLOCKED
65
69.364
0.022
9.81
45.076
11.382
3.074
45.05
333.08

0A 01 1C 02 01 10 00 93
30 26 A1 42 1F 58 00 00 94 94 32 61 23 42 01 1B C1 D0 8E 65 90 BE 00 43 F1 02 0E C1 00 A1 00 87