mnh-jansson/open-battery-information #41: BL1850B with charger slow flashing red light

Orgman67, 2025-02-18:
Hi I have a BL1850B battery and when I put on the charger it starts to charge then after a few seconds stops and the charger displays a slow flashing red light. This is indicating a temperature issue.

Does anyone have any thoughts on what might be the problem with the board as the thermistor appears to be working ok. Also where is the 2nd temperature sensor located ?

[image]

[image]

Alexys-seven, 2025-02-18:
Hi, I have the same situation with one battery, please check my post in discussion section "cell voltage difference". Martin has explained that voltage difference readings is not correct, but your battery has the same difference.

Second thermistor is inside mosfet under black coating.

Bennypro66, 2025-02-18:
Hello, I have the same problem. The fuses are good and the balance is perfect and the temperature sensor is good. And yet like you it indicates a voltage difference of 0.28××× while everything is good. I think a resistor must be damaged on the bms card.

[image]

Bonjour, j'ai le même problème. Les fusible sont Bon et l'équilibrage est parfait et la sonde de température est bonne.  Et pourtant comme vous il m'indique un une différence de voltage  de 0.28××× alors que tout est bon.  Je pense qu'une résistance doit être endommagé sur la carte bms.

ch9511, 2025-02-19:
0.2xx voltage difference is standard for Taiwan made Makita batteries with Samsung cells.There I think is thermistor issue try to change it and reset battery errors.Also you can check cell voltages on 10A load, if some of cells lost capacity you can see that.

Alexys-seven, 2025-03-04:
> Hello, I have the same problem. The fuses are good and the balance is perfect and the temperature sensor is good. And yet like you it indicates a voltage difference of 0.28××× while everything is [...]

I have one battery where 150 ohm resistor was damaged, there was big voltage difference in readings.
 [image]