mnh-jansson/open-battery-information #43: Unlocked battery with arduino but cannot charge

BF1185, 2025-02-20:
The program detects my battery normally.
I read the battery, it says that the battery is locked (State: Locked).
The light test in the program works, by clicking on Clear Error, it unlocks the battery.
The battery works on the machine, but it cannot be charged.
The red and green lights on the charger are flashing.
I have tested four different batteries (3ah, 5ah and 2x 6ah) and they all have the same problem.

Please for the solution?

Thank you.

no-body-in-particular, 2025-02-20:
Please show us some more information from OBI. The PCB could be broken for instance, how is the balancing on the cells etc? If you do not resolve the cause of the error code the charger will always just block the battery again.

BF1185, 2025-02-20:
Cells have a very small deviation of 0.02
I doubt that the electronics on all four batteries are faulty. I will send more data from OBI in the next few days.

ch9511, 2025-02-21:
Can be faulty charger, or mud on connector.

BF1185, 2025-04-10:
I finally found time to continue...
I sending information from OBI:

Baezery Data:
BL1830B
311
UNLOCKED
67
17.455
3.498
3.494
3.497
3.494
3.495
0.06208
29.72
29.86
12 09 09 64 14 08 10 57
F1 26 BD 13 14 58 00 00 44 44 40 21 D0 80 02 1F E1 D0 8E 67 60 FE 00 61 02 02 1E 73 10 40 00 27

[image]

Debug information:
Imported module: makita_lxt
Imported interface: arduino_obi
>> 01
<<
>> 01
<<
>> 01
<< 00 02 01
Opened serial port: COM4
>> 33 AA 00
<< 12 09 09 64 14 08 10 57 F1 26 BD 13 14 58 00 00 44 44 40 21 D0 80 02 1F E1 D0 8E 67 60 FE 00 61 02 02 1E 73 10 40 00 27
>> CC DC 0C
<< 42 4C 31 38 33 30 42 00 00 00 00 00 00 00 00 00
>> CC D7 00 00 FF
<< 2F 44 AA 0D A6 0D A9 0D A6 0D A7 0D 40 18 9C 0B AA 0B 00 80 40 24 11 23 09 40 9F 11 00

BF1185, 2025-04-11:
Battery working after unlock...but it can't be charged....I've tried multiple batteries and multiple chargers and is the same....please for help.

ch9511, 2025-04-11:
Where these batteries was made,in China or Taiwan? They use different BMS boards.Can you open it and take picture of circuit board?

BF1185, 2025-04-11:
No...BMS boards are not cheap Chinese...
The program would not recognize chinese bms.
Electronics are definitely ok...batteries are also without defects....

ch9511, 2025-04-11:
I’m not talking about cheap Chinese batteries.It is original Makita batteries made in China with Sony Murata cells and BMS with numbers LIPW0xx and Makita batteries made in Taiwan with Samsung cells and some another type BMS where is not MAK0x chip.Program can read  data from both,  but I’m not shure at program can unlock it.

BF1185, 2025-04-11:
But i have 5 different batteries...is not possible that i all from china... the problem is probably somewhere else 😕

BF1185, 2025-04-12:
no no. I'm familiar with it. .tried several chargers... as I said the tool works with an unlocked battery..it just can't be charged. 
The charger is not charging this battery..the red and green lights are flashing..

mnh-jansson, 2025-04-12:
Please clarify what the charger does when you place the battery on the charger.

1. Does the battery instantly go to red/green blinking?
2. Does the red light stay on for a little while and only then go to blinking?

BF1185, 2025-04-12:
Yes point 2...stay red for a little and then blink...

mnh-jansson, 2025-04-12:
Ok, this is typical of blown fuse.

BF1185, 2025-04-12:
Fuse on battery bms?

mnh-jansson, 2025-04-12:
Yes

BF1185, 2025-04-12:
But battery working normaly on tool when i unlock with you program... i trying on 4 different battery 3ah, 4ah 
5ah, 6ah and on all is the same result.. where is this fuse? Location on Bms?

mnh-jansson, 2025-04-12:
The fuse location depends on the board. 

[image]

Most Makita batteries have the fuse on the charge pin of the connector, so it won't affect the tool.

BF1185, 2025-04-12:
I'll check this fuse on one of the batteries and let you know...

BF1185, 2025-04-13:
Battery is made in Singapore - 6ah. This is board..Where is location of this fuse on this board?

[image]

mnh-jansson, 2025-04-13:
[image]

BF1185, 2025-04-13:
ok. I thought so...I'll try a battery with a good fuse tomorrow...

mnh-jansson, 2025-04-13:
I can't of course recommend bridging the fuse, but it's always a possibility :)

BF1185, 2025-04-16:
Yes..You are right..The fuse blew...Thanks for your help and effort. You did a great job on this project!