mnh-jansson/open-battery-information #44: [QUESTION] MacOS build

martin-badin, 2025-03-01:
Hello.

First, thank you for this project. Can I run this app on MacOS?

yukonx, 2025-03-02:
On macOS Apple M4 (15.3.1), the program can run, but I am unable to drive the COM port. On macOS 13.7.4 (Intel), it works fine.

martin-badin, 2025-03-02:
Please. How to do it? I have an Intel mac. Or it is possible to includes mac os app in releases

mnh-jansson, 2025-03-02:
You will need to install Python, but it should work on Mac as well. I don't have a Mac so i cant help with this, but i would assume googling how to install Python on Mac would give quite a few results.

jimmykl, 2025-03-02:
Not sure if Intel/Apple Silicon should make any difference as packages are available for both architectures but I can't test this.
This works for me on my M1 Max macOS 15.3.1

Install [Homebrew](https://brew.sh/)

In terminal, install python

`brew install python`

Get the code (you can also just download if you don't care for git)

`git clone https://github.com/mnh-jansson/open-battery-information.git`

Setup and activate a python virtual environment (maybe optional but I always do this to avoid package conflicts)
```
cd open-battery-information
python -m venv ./obi-venv
source ./obi-venv/bin/activate
```
Install packages required

```
cd OpenBatteryInformation
pip install -r requirements.txt
```
Run it

`python main.py`

yukonx, 2025-03-02:
On macOS, you need to run the following commands:

python3 main.py
pip3 install -r requirements.txt

These commands execute correctly.

yukonx, 2025-03-02:
In macOS dark mode, the display appears abnormal.

[image]
[image]

Snellenburger, 2025-03-05:
I am also having trouble entering the serial port  I have downloaded Arduino, downloaded python. followed YouTube videos and seem to be stuck on the last bit.  any help would be much appreciated . 

im not good at programming or things like that. has anyone seen any Mac videos I could follow ? 

[image]

martin-badin, 2025-03-15:
I created PR

https://github.com/mnh-jansson/open-battery-information/pull/47