mnh-jansson/open-battery-information #46: BL1830

Adnanzo, 2025-03-15:
-my battery case says BL1830 but inside looks like this , my camera could not catch the mcu but i clearly saw F0513 on the square shaped chip
it is locked as it does not charge and red flashes and beebs none stop
-did anyone have any progress in including it in the final code so we can reset errors and unlock bms
as this pack is old but good sony cells 1500mah in it , charged manualy and balanced and can run the drill if inserted midway before it locks in place , discharges with an led load fine
should i desolder the bms and see if it unlocks as i read that this might work.
thank you all.

[image]

TBAMax, 2025-03-15:
@Adnanzo  When it works only half way inserted, this means ou probably have this kind of drill that uses the third contact. The third contact is used for low battery protection on newer drills. On older batteries like yours this third contact simply is connected to the "batt+" via the heated fuse. This it kind of fuse that can be blown by the software using the built in heater. So the symptom indicates that this fuse is physically blown (it can not be recovered by software), you would need to dissassemble the fuse and resolder the solder bridge that was melted. Though this alone probably will not fix the charging, because the processor still has some kind of error code inside it.

Adnanzo, 2025-03-15:
exuse me sir but by fuse do you mean the heat sensor resistor , it gives 11k ohms when tested at 25 celuis.
-if not then can the fuse be fixed on it's own , or is it inside a chip , could you maybe point the way to it.
-as for error reset it should be possible after the fuse is fixed or bypassed right.
thank you for you detailed respectful answer.

TBAMax, 2025-03-15:
> exuse me sir but by fuse do you mean the heat sensor resistor , it gives 11k ohms when tested at 25 celuis.
-if not then can the fuse be fixed on it's own , or is it inside a chip , could you maybe point the way to it.
-as for error reset it should be possible after the fuse is fixed or bypassed right.
thank you for you detailed respectful answer.

Look I did put an arrow where the fuse is. You can pry the lid off of it and fix it with the soldering iron.

[image]

nlutterman, 2025-06-02:
Thanks for sharing that, @TBAMax, very helpful. To make sure I understand you right: you're saying there's likely a flag set in ROM somewhere when the fuse blew, indicating a malfunctioning battery.

It's a bit of a reach, but would you happen to be able to direct us to any information about the board? I wonder if it'd be able to wire up some debugging tools to reset that flag. As long as I can get pin out info and maybe memory layout information, I'd be able to go the rest of the way myself.

Thanks again!

mnh-jansson, 2025-06-02:
Most common reason for fuse to blow is that the microcontroller has detected a fault. This fault is usually that a cell connection has broke, causing the BMS IC to think there is a high cell voltage and triggering the fuse to blow. This will be recorded into the emulated EEPROM in the case of F0513.

F0513 does not have a command to clear this flag as far as i am aware. There is no debugger in F0513A, not even a function to read memory. I do have a workaround, but this is not public yet.

Adnanzo, 2025-06-08:
thanks , i managed to rig drill to accept voltage from battery pack with
board as is by connecting middle pin "so middle pin no longer connects to
board when battery is inserted" of drill to volt + of drill , that way i
tricked the battery pack and drill to think the board works fine , but the
battery discharges not balanced at first use it manged to do 13 holes , i
hardly use the drill so limiting the use to 4 holes works , did not use
original charger to fill it up , used an other lithium charger , will buy a
new board and build this battery soon or buy new pack.
not giving up on original board due , cheaper to fix it to accept new cells.
keep up the good work my friend .