mnh-jansson/open-battery-information #48: [Question regarding alternative battery]

rodrigolange, 2025-03-24:
Hello!

I have a question regarding those Makita battery signals.

I have a Makita Coffee Maker, and want to power it with my car 12V power source (or with any power source).

The coffee Maker accepts BL1041b 12V batteries, but as any new Makita batteries, they must answer queries from the device.

Do anyone have any ideia on how can I emulate the BL1041b and use a regular power source or another generic 12V battery?

Alternatively, I'm thinking of simply buy a BL1041b and replace the internal batteries with a power cord to my power source. Will it work?

Thanks in advance,

Rodrigo

no-body-in-particular, 2025-03-25:
Quickest solution, buy this PCB https://nl.aliexpress.com/item/1005006478165394.html?gatewayAdapt=glo2nld put a 1k resistor in place of each battery cell as on schematic, connect the + and - to the PCB. 

Alternative ( non-digital ) : a coffee maker is just a heating element, with an overtemperature and sometimes water level switch. Connect 12v directly to the heating element, and put a relay connected to the overtemperature cutoff inline.

rodrigolange, 2025-03-25:
Thanks!

I didn't want to use the second solution to avoid messing too much with the
coffee maker, although I admit it is the simplest solution.

Will think about it while drinking coffee. :-)

Thanks!