mnh-jansson/open-battery-information #49: Як запрограмувати Arduino nano

Vlad-Pivtorak, 2025-04-02:
I can't suggest an Arduino Nano.

CTRL-ALT-UP, 2025-04-03:
I've used a nano with no issues so far.

yukonx, 2025-04-04:
Follow this YouTube tutorial, and you can complete it: https://www.youtube.com/@allans-workshop/videos

[image]

[image]

kais1988, 2025-06-20:
Видео не открывается.

saulega2, 2025-11-21:
Kur galima isigyti toki daigta?

kolaha99, 2026-05-30:
> I can't suggest an Arduino Nano.

it is not clear what you mean, do you not recommend to use genuine Arduino Nano or you are not able to upload the program in your Arduino Nano for some reason? Is your problem still actual?