mnh-jansson/open-battery-information #51: "OBI" program writes the message "not supported battery version"

mva-ru, 2025-04-15:
I managed to connect to my Makita battery. But your OBI program gives an error message "unsupported battery version". And accordingly I can't reset the error, the reset buttons in the program are not active. But at the same time, the data about the battery and the battery voltage are successfully read.

My Makita battery does not have a capacity indicator and was bought quite a long time ago, more than 5 years ago.

I am attaching a photo of the battery and the logs read from it (battery ID - 0B 08 09 02 03 1A 09 03, model - bl1830). 

Please add my battery to the support of your program so that I can reset the error.

[image]
[log_makita_bl1830.txt](https://github.com/user-attachments/files/19751866/log_makita_bl1830.txt)
[image]
[image]
[image]

Thanks in advance.

Thank you.

mnh-jansson, 2025-04-15:
The warning message is not "unsupported battery version", it's "This battery only supports diagnostics". The reason for this warning is exactly what it says, it only supports diagnostics. There is no functionality in this BMS that resets the error flags. I have a workaround that can be used to solve this, but this solution is not yet opensourced.

Also, your battery is from 2011, so 14 years 😉

eroyee, 2025-04-15:
> ... I am an embedded systems programmer, maybe I can solve this problem myself....

Hopefully you + Martin and others with the skills could solve the problem collaboratively on here so that everyone is able to benefit?

While I am fortunate in not having the issue you have I do applaud Martin's initiative in establishing an **open** community / knowledge around the otherwise closed world of Makita. _Un_fortunately my skills in this area are limited but I would like to record encouragement that any developments do occur openly and freely - which in my experience tends to bring people together and can help things progress well.

To avoid doubt - I'm absolutely not getting at anyone here, just wanted to provide some positive words about working together and seeing things happen here :)

mnh-jansson, 2025-04-15:
Like i mentioned i do have a workaround. Unfortunately this involves a method that i prefer not to release into the public, alteast for now. I can offer to unlock over remote desktop if you want, in that case contact me over email.

Alexys-seven, 2025-04-15:
I have managed to owervrite battery message, battery become unlocked in OBI, but it does not charge in genuine charger, red led blinking(not red green). Fuse is ok, thermistor ok too, battery cell are balanced. Have not more ideas what is wrong.

mva-ru, 2025-04-15:
It is not in my rules to give remote access to a PC to a stranger.

I decided to replace the board and buy it on Aliexpress ($3):

https://aliexpress.ru/item/4001015685900.html?sku_id=10000013516465351&spm=a2g2w.productlist.search_results.0.59c25220aJDuS3

eroyee, 2025-04-15:
> a method that i prefer not to release into the public

I'm sure you have your reasons but I have to say this seems unusual from what I understand 'open' to mean, even more so given the reason for this project to exist? 

Oh well, I suppose one creates less e-waste than throwing out the whole battery by simply replacing the board as @mva-ru has pointed out - although with all this hostility in the world who knows if/when that option becomes unavailable :(

Otherwise, in the interests of open discussion on this, has anyone tried 'tricking' the charger into thinking it has a 'good' battery attached to it - or somehow modifying the charger system? I don't have the issue so can't exactly try anything, but I guess it might be another avenue of research. Would be interesting to know if a battery that a _Makita_ charger refuses to charge, and/or is otherwise 'locked', would record further charge/discharge cycles when charged in another way.

mnh-jansson, 2025-04-15:
@mva-ru 

And that's completely fine by me :)

mnh-jansson, 2025-04-15:
@eroyee 

"open" has to have some limits when working with others proprietary information.

The method i mention enables the possibility to read the full firmware from the BMS. It uses a bootloader that is included in the firmware to first read the memory, parse the memory structure, clear the error flags and finally erase and rewrite memory. 

Marking this issue as closed now as it is already a duplicate issue and the author has gotten his answer.

closed 2025-04-15 by mnh-jansson

TBAMax, 2025-04-15:
> The method i mention enables the possibility to read the full firmware from the BMS. It uses a bootloader that is included in the firmware to first read the memory, parse the memory structure, clear the error flags and finally erase and rewrite memory.

Well, that is advanced. Since this board has the F0513 processor, which is known to be locked, and the firmware extraction is quite an advanced procedure, as we have read from @mnh-jansson's blog posts...
So, reading the important parts of the firmware (mentioned flags and possibly the calibration data) using the Arduino and the onboard bootloader is even more advanced.

I cannot see exactly what the "legal" problems would be in this case... 14-year-old tech. Anyway, it's enjoyable reading.

mnh-jansson, 2025-04-15:
Unless you're a lawyer i'm not gonna take your word for it 😅 Thanks

eroyee, 2025-04-18:
@mnh-jansson 

Thanks for the explanation, it helps to understand the reticence. 

FWIW my view is that practically you/we are simply wanting to repair our batteries that we have purchased and which through a deliberate manufacturing defect are liable to be rendered useless - which in turn causes damage to us, financially at least. 

We _intend no harm_ and wish simply to repair and/or return to service said batteries. In many jurisdictions reverse engineering is not a crime, particularly when it is to correct a wrong and/or not for the purpose of unduly profiting from the work of another person. 

Also, as @TBAMax has noted, this is aged tech and there are existing workarounds that have not attracted any action that we know of (or has not been successful, as the workarounds remain).

Thus I don't see any issue, although to be clear this is merely a comment in discussion, not an advocation to you...

keyang903, 2025-11-21:
@mnh-jansson I also have two BL1830B battery packs that are displaying the warning message \"Unsupported battery version\", \"This battery only supports diagnostics\". I also hope you can help unlock them. Thank you