mnh-jansson/open-battery-information #55: How do I lock the battery?

Fanzetess, 2025-05-04:
The battery unlocks after running the program, but I need for my tests, it also locks. How can I lock the battery?

ch9511, 2025-05-06:
Just cut one wire from the cell and put in on charger or discharge one cell to get misbalance and battery will be locked.

Bennypro66, 2025-05-06:
Salit
Attention si tu coupe le lien entre une celule et la carte, ça va bien faire une erreurs. Mai ça risque de faire sauter un fusible de sécurité quand tu la branche au chargeur

Bennypro66, 2025-05-06:
Si tu débranché le + de la dernière celule, et que tu met ensuite sur le chargeur.. le fusible devrais rester intact
Et ça devrait crée une erreur 
[image]

Fanzetess, 2025-05-06:
Is there any way to lock it programmatically? Using the same program, but adding additional code?

Fanzetess, 2025-05-06:
> Makita BPS01

What?