mnh-jansson/open-battery-information #58: failed to connect

SoeWinHtun, 2025-06-02:
[image]

I can't connect with my UNO. " Error opening serial port "
Somebody help me please

SoeWinHtun, 2025-06-04:
[image]
Imported module: makita_lxt
Imported interface: arduino_obi
>> 01
<< 
>> 01
<< 
>> 01
<< 00 02 01
Opened serial port: COM13
>> 33 AA 00
<< 15 08 08 64 14 08 10 F3 F1 26 BD 13 14 58 00 00 B1 B1 40 21 D0 80 02 5E C3 D0 8E 67 9F 16 00 41 02 02 0E 31 00 D0 04 75
>> CC DC 0C
<< 42 4C 31 38 36 30 42 00 00 00 00 00 00 00 00 00
>> CC D7 00 00 FF
<< 6F 4A 55 0F B4 0E B7 0E B4 0E F9 0E 40 18 ED 0B EB 0B 00 80 40 5C 38 03 16 99 9C 8B 00
>> 33 D9 96 A5
<< 15 08 08 64 14 08 10 F3 06
>> 33 DA 04
<< 15 08 08 64 14 08 10 F3 06
>> 33 D9 96 A5
<< 15 08 08 64 14 08 10 F3 06
>> CC F0 00
<< F1 26 BD 13 14 58 00 00 94 94 40 21 D0 80 02 5E C3 D0 8E 67 60 FB 00 41 02 02 0E 31 00 D0 04 75
>> 33 33 0F 00 F1 26 BD 13 14 58 00 00 94 94 40 21 D0 80 02 4E 23 D0 8E 45 60 1A 00 03 02 02 0E 20 00 30 01 83
<< 
>> 33 55 A5
<<

SoeWinHtun, 2025-06-04:
I can't unlock. Please Help.

Bennypro66, 2025-06-05:
[image]
Vérifier toute les connection en nickel relier au celules au bout des flèche et le fusible que j'ai entouré