mnh-jansson/open-battery-information #59: Problem with unlocking 6Ah battery

BF1185, 2025-06-11:
[image]

Hello,

I have three batteries with the same problem... I checked everything on the batteries.., is there a broken contact or fuse... but I can't UNLOCK the battery.Has anyone had a similar case?
I have one battery that only works on tools with two contacts on the terminal (radio, vacuum cleaner,...) .. When i insert it into a tool with three contacts on terminal...Then the battery locks immediately.  What can I do? Thanks.

ropewo, 2025-06-12:
That board looks very similar to my 1850B which has the F0513 BMS. Do you get the message "This model only supports diagnostics"? In that case the current version of the OBI-program doesn't support unlocking.
I also have a couple of BL1815N which is supported. I can read model and data even if the cells ar not connected. The reset buttons in the program are active but the BMS remains locked if I try a reset.

I would first make sure I have a supported BMS and then try to charge the cells individually.

BF1185, 2025-06-12:
I connect to the battery without any problems... I see all the data of battery... I also turn the lights on and off... I can't unlock it...

ch9511, 2025-06-12:
Lipw014 board is fully supported. Battery cut off voltage above 12,5v try to charge it to 14-15v with power supply or from another battery. If it not help change fuse and check current resistors on back side of board

WilsonTakeshi, 2025-06-17:
Olá Sergiy9019, o carregador que usou foi DC18RC ou DC18RF? Ouvir dizer que o carregador DC18RC tem a capacidade de resetar a bateria caso a falha da bateria não seja muito grande.

cgm999, 2025-10-20:
I have a BL1820 which only supports Diags.. and is locked. Delta between cells is very small.. Any chance to unlock it ?

ROM ID: 15 06 08 02 06 33 09 E9

Debug line:
D1 22 BD B3 16 58 00 00 B1 B1 40 61 B2 81 01 01 41 D0 8E 67 3F B4 00 03 02 02 0E 20 00 10 02 73

tyvm!

juppp, 2025-12-16:
Hey @ch9511, regarding the fuses: I measured a SEFUSE® D6SC5 15A and it is reading Open (no continuity) between Pins 1 and 3. This is not what I expected?

[image]

[image]

ch9511, 2025-12-16:
If it’s open circuit between 1 and 3 fuse is blown . Just solder wire between 1 and 3 if charger will start charge, change fuse.

ch9511, 2026-02-24:
Just solder wire on top.And check fuse. It’s common problem with BL1860B

MeatEasterPearls, 2026-02-24:
you were right, that was the problem. Thank you!