mnh-jansson/open-battery-information #60: Charger problem! BL1840B pack with 6.0Ah capacity (new better cells)

BossePedal, 2025-06-14:
Is it possible to rewrite the "type of battery" in the BMS?

I have swapped the old bad cells in my 4Ah pack to new higher rated cells that gives the pack 6Ah.

Everything works, but the charger seems to expect the lower capacity cells and terminates the charge prematurely. If the charge starts at low SoC the charger stops at ~19,5V and plays the tune. If I then restart the charge it continues to the normal ~20,5V. It feels like the charger have an[image] mAh counter and a max value it will not go beyond to prevent overcharging. I think this is a very good safety mechanism, but annoying for me with my new higher capacity cells.

Is there a way to make the BMS think it is a BL1860B pack?

WilsonTakeshi, 2025-06-17:
Você pode estar com um par de baterias com tensão mais alta que as outras do pacote. Equalize as tensões uma a uma com uma fonte de DC 4,2V ou aguarde alguns dias que a BMS Makita tem a capacidade de equalizar as tensões automaticamente. No final, pode tentar carregar a bateria. Faço isso com as minhas baterias e raramente não dão certo. Boa sorte!

mnh-jansson, 2025-06-17:
There is a milliamp register in the battery message, this might be used by the charger as a limit. Would be interesting to experiment with this.

danyaPostfactum, 2025-06-22:
I would like to know solution also. But i use "dumb" chinese small "BMS" for 9Ah battery. Anyway.
I remember i saw some chinese battery kits with "charge restart" advertisement (capacity limit workaround). So, i guess, capacity limit is hardcoded inside a charger...

It would be nice to have some open source projects of custom charger firmware 🤤.

Alexys-seven, 2025-06-23:
It is possible to owerwrite battery message, but it has checksum. At this moment I'm not sure how to calculate it

BossePedal, 2025-06-23:
Please let me know if you find out how to calculate the checksum. Thanks 😊

mnh-jansson, 2025-06-23:
I have the checksum figured out, but i have not tested the effect of changing the mAh bytes yet.

mnh-jansson, 2025-06-25:
Maybe you could contact me through email and we can try set something up?

BossePedal, 2025-06-25:
Yes, I'll send you an email. Will do that in beginning of next week. I'm going on vacation now. Talk soon

Alexys-seven, 2025-06-28:
I have checked it on several batteries and looks like checksums are at nybbles: 
40 (sum of nybbles 0..15)
42 (sum of nybbles 32..39)
43 (sum of nybbles 16..31)

Hope this will help you in your experiments and you will share results