mnh-jansson/open-battery-information #66: BL1840B Vietnam Samsung lock after unlock

Sergiy9019, 2025-07-02:
Hi
I am trying to unlock bl1840b battery Vietnam assembly. It was locked by non original charger (21,3 V definitely not a normal end of charge voltage).
It reads by OBI well, after pressing “Clear errors” it unlocks, works good with tool, charges on orig charger dc18wc well, but when I take the battery from charger, that moment it locks again and starts blinking pair indicators.
Everything seems to be good: cells are good, balance perfect, thermistors sefuse good, all connections battery pack good.
Tried even disconnect the board from battery pack from + to -, but the effect is the same.
What can it be else? Or maybe it is something like permanent lock in controller memory?
Controller is 240080

[image]
[image]

Sergiy9019, 2025-07-03:
One more interesting thing on this.
In OBI when press “clear errors” then press “read battery model” - it unlocks, then press “reset battery message” - “read battery model” - it locks back. That moment I hear something like “click” sound from the BMS, but I don’t understand what can do that sound. Maybe mosfets blocks? What else can it be?

Alexys-seven, 2025-07-03:
Looks like your battery doesn't like store command 0x33 0x55 0xA5. If you have not modified OBI code, reset battery message function does not work, but it sends store command, you can see it in debug window. Charger sends the same command at end of charge cycle to rewrite battery message (increase charge count and maybe something else). I have noticed that this command itself sometime overwrites some bytes of battery message, but I don't know why.

mnh-jansson, 2025-07-07:
I've experienced the same issue, but with LIPW017 5Ah. Have not found what the issue could be, but seems to be hardware related.

Sergiy9019, 2025-07-09:
So, after my some manipulations with the that board, MCU died. I don’t really understand how, I just suspected the thermistor rattling connection and resolder it. Then the board did not response on anything and MCU was heating up. The measurement showed that MCU internal 3,3 v regulator is shorted inside now…
So I have replaced the Vietnam board with lipw014 one, that I have bought online.
Now it looks like this

[image]

It was a really hard work to get that together. Vietnam and China batteries are completely different inside.
But it works fine now.
lipw014 board unlocked easily with OBI after installation.