mnh-jansson/open-battery-information #71: BL1830B Cell 5 reports 0 but ok on muiltimeter

leed71, 2025-07-07:
Hi 

I know this isn't an issue with the code, just wondering if anyone had any ideas why its reporting zero for cell 5. When I measure all the cells they're reporting ~3.79V and the whole pack is @ 18.9V I tried reflowing all the connectors to the board but that didn't help.

[image]

[image]

mnh-jansson, 2025-07-07:
There should be a resistor in series with the cell connection that goes to the BMS IC, make sure that is still ok. I think that all boards have 150 Ohm resistors.

leed71, 2025-07-08:
Has anyone got any top tips on removing the goo?

leed71, 2025-07-08:
Well that was a yin and yang day, during a very boring meeting, I very slowly peeled off the goo with some IPA

[image]

Alas at some point some components seem to have gone missing, even thou I only used my finger nail

[image]

Guessing this board is now toast, unless anyone has any ideas what they were? SMD's are new to me.

I also didn't see any 150Ω resistors (the code is 151?) But as you may have guessed I'm a bit of a newb.

Also I realise the issue section isn't to correct place to talk about this unless Martin is happy with this, more than happy to take it elsewhere (where ever that is)