mnh-jansson/open-battery-information #80: Troubleshooting Wiki

qwertymodo, 2025-09-03:
As more people start getting into this, chances are they will tend to run into similar problems over and over again.  For instance, I see a lot of references in threads here about checking the fuses, or the cell resistors.  Rather than repeatedly answering the same questions over and over, it would be really helpful to document various symptoms and their causes/solutions all in one place.  I see you already using the wiki feature here to document various BMS board revisions, it would be nice to add pages related to the troubleshooting and repair process, that way we can start to build a knowledge base for repairing actual physical problems with batteries beyond just resetting the BMS error flags.

mnh-jansson, 2025-09-04:
Definitely. I have mentioned this in the Discussions section, but i currently don't have the time to do it. The wiki should be accessible for anyone to edit i think, maybe i'm wrong?

ropewo, 2025-09-19:
@mnh-jansson I'm logged in and I can't edit the Wiki.
[Changing access permissions for wikis](https://docs.github.com/en/communities/documenting-your-project-with-wikis/changing-access-permissions-for-wikis)